xoseperez/espurna #1103: SoftAp default name to "ESP"-DEVICE-xxxxxx

hyteoo, 2018-08-04:
The current name of the AP when espurna is not connected to WiFi is "ESP_xxxxxx" which is very confusing when you have several devices running

I propose we change that to "ESP-" + DEVICE + xxxxxx so it is clearer which device is which.

I tried modifying this in utils.ino in order to do a PR, however it seems that the modifications are ignored. I even set a hardcoded string in wifi.ino, no luck.

Also the AP name is "ESP_" with underscore, but in utils.ino it is specified with "-".

Where is this name overriden?

xoseperez, 2018-08-13:
The actual name it uses is the one stored in the `hostname` config variable. Only if there is no `hostname` configured it gets the value from the `getIdentifier` method. So you might want to check if you have that variable defined and delete it (`del hostname` from a telnet session or in the web debug tab).

On the other hand, the name it creates should be "ESPURNA-XXXXXX", not "ESP-XXXXXX".

hyteoo, 2018-08-17:
Hi Xose, sorry for the delay, vacation happened.
I just turned off my APs on which all ESPs connect to, so they will all switch to AP mode.
Here's a sample fo what I get in Wifi APs:
[image]
Weird thing is that one of them is ITEAD...
Obviously, all devices have a hostname assigned.
It's like the AP name is stored somewhere and the name set by espurna is ignored.

hyteoo, 2018-08-22:
Small update: I found this in another ticket, not related.
How the AP names look:
[image]
I have yet to find out why this is happening

mcspr, 2018-08-22:
@hyteoo `ESP_...` comes from [justwifi lib](https://github.com/xoseperez/justwifi): https://github.com/xoseperez/justwifi/blob/master/src/JustWifi.cpp#L52
If no SSID provided, it uses hostname: https://github.com/xoseperez/justwifi/blob/master/src/JustWifi.cpp#L296-298

As Xose did mention, something funky must be happening to the settings. Only thing that makes sense if hostname is empty here / `hostname` key from eeprom settings returns nothing.
https://github.com/xoseperez/espurna/blob/4f61d3aa7e2e6526fdb518a9f84785b055ac0ab7/code/espurna/wifi.ino#L36-L40

And looking at wifi listings only common thing I did notice that in all cases ESP APs are passwordless, but i do not know if this is relevant to this issue.

hyteoo, 2018-08-23:
@mcspr then maybe we should create a different issue for this, and keep this one as a feature request.

hyteoo, 2018-08-23:
@xoseperez after some more tests, I think we should keep the original naming scheme (at least as it is in the code: ESPURNA-xxxxxx)
Seems the longer name causes problems especially in MQTT client name.
I suggest we close this one.