xoseperez/espurna #1104: h801 restarting with latest dev build

shaguarger, 2018-08-04:
When flashing the latest dev build (commit f18c901c3f0f99211175c08a5b204311609f4447), the LED stripes controlled by the h801 are flashing with the highest brightness after some period of time and the h801 is restarting. This does not happen with the latest master build.

The restart was reproducible on 4 h801 when setting the following values:
State: ON
Channel #3 1
Channel #4 255

There's unfortunatley no log message in the DEBUG log screen, however here is it, just in case:

```
[027055] [WEBSOCKET] #1 connected, ip: 192.168.178.22, url: /ws
[038982] [MQTT] Received wohnzimmer/sued/relay/0/set => 1
[038986] [RELAY] #0 scheduled ON in 0 ms
[038989] [RELAY] #0 set to ON
[038991] [MQTT] Sending wohnzimmer/sued/rgb => #FFFF00 (PID 1)
[038993] [MQTT] Sending wohnzimmer/sued/hsv => 0,0,0 (PID 1)
[038996] [MQTT] Received wohnzimmer/sued/brightness/set => 0
[039001] [MQTT] Sending wohnzimmer/sued/rgb => #FFFF00 (PID 1)
[039005] [MQTT] Sending wohnzimmer/sued/hsv => 0,0,0 (PID 1)
[039010] [MQTT] Sending wohnzimmer/sued/brightness => 0 (PID 1)
[039016] [MQTT] Sending wohnzimmer/sued/mired => 326 (PID 1)
[039021] [MQTT] Sending wohnzimmer/sued/channel/0 => 255 (PID 1)
[039027] [MQTT] Sending wohnzimmer/sued/channel/1 => 255 (PID 1)
[039032] [MQTT] Sending wohnzimmer/sued/channel/2 => 0 (PID 1)
[039038] [MQTT] Sending wohnzimmer/sued/channel/3 => 1 (PID 1)
[039043] [MQTT] Sending wohnzimmer/sued/channel/4 => 255 (PID 1)
[039052] [MQTT] Received wohnzimmer/sued/channel/3/set => 1
[039055] [MQTT] Sending wohnzimmer/sued/rgb => #FFFF00 (PID 1)
[039060] [MQTT] Sending wohnzimmer/sued/hsv => 0,0,0 (PID 1)
[039065] [MQTT] Sending wohnzimmer/sued/brightness => 0 (PID 1)
[039070] [MQTT] Sending wohnzimmer/sued/mired => 326 (PID 1)
[039076] [MQTT] Sending wohnzimmer/sued/channel/0 => 255 (PID 1)
[039081] [MQTT] Sending wohnzimmer/sued/channel/1 => 255 (PID 1)
[039087] [MQTT] Sending wohnzimmer/sued/channel/2 => 0 (PID 1)
[039093] [MQTT] Sending wohnzimmer/sued/channel/3 => 1 (PID 1)
[040148] [RELAY] Saving mask: 1
```

shaguarger, 2018-08-11:
We now found out, that this also happens with the current master...

mcspr, 2019-11-09:
Closing due to the age. I'd think of this as a prominent power supply issue or something that was already (hopefully) fixed with the current version.

closed 2019-11-09 by mcspr