xoseperez/espurna #1105: New device:

hyteoo, 2018-08-05:
I got these from Amazon: https://www.amazon.de/gp/product/B07D7RVF56/ref=oh_aui_detailpage_o00_s01?ie=UTF8&psc=1
[image]
Nice little sockets, they have the following features:
- extremely compact
- 16A relay
- 4x RGB LEDs
- energy monitoring with HLW8012

Tested them with espurna, created a new hardware profile, so far they work great.
I'll submit a PR shortly

There are 3 boards, sandwiched together like a H.
The outer ring is illuminated by the 4 RGB LEDs, making the whole socket look very nice. Maybe some PWM would work nice here.
Here are some internal shots:
![20180804_184041](https://user-images.githubusercontent.com/1378206/43683869-c2a8d9e0-989d-
11e8-87b6-d6e26d4aba23.jpg)
[image]
[image]
[image]
[image]
[image]
[image]

xoseperez, 2018-08-05:
Merged! I'm sure this is a great device.

flashy02, 2018-08-09:
One question: How can I open this Device without damaging?

hyteoo, 2018-08-09:
Just pop out the plastic ring with a cutter or a plastic pry tool. It should pop without too much effort.
There are no clips, just a small amount of glue.

flashy02, 2018-08-10:
Thank you very much. The opening and flashing worked just fine but the first installation was a mess. Please look issue #1119

hyteoo, 2018-08-10:
Hmm that is weird, I had 2 homecubes flashed with eapurna without issues.

flashy02, 2018-08-10:
Perhaps my device has a problem. But at the moment everything works fine. Only the first installation was difficult. Which version of espurna have you flashed?

hyteoo, 2018-08-11:
Latest master branch

flashy02, 2018-08-11:
The same as I have used. I would try it again with an other homecube/device, soon. Perhaps it is a thing of this special device.

hyteoo, 2018-08-23:
@ruimarinho I'm not sure what you mean
The usual suspects: Tx,Rx, GPIO0 for flashmode.
And the pinout
[image]

andreasloe, 2018-08-24:
Regarding flashing: I do not see this chip on the picture above. Also, the pictures show the ESP8266, but you mention  WT8266-S1. Thanks in advance!

hyteoo, 2018-08-24:
The picture above is the pinout
I never said the device has the [WT8266-S1](https://www.seeedstudio.com/WT8266-S1-FCC%26amp%3BCE%26amp%3BRoHS%26amp%3B-Wi-Fi-Module-based-on-ESP8266-p-2521.html), just that the pinout is the same. Even though it also has an ESP8266 inside, just rebranded the entire assembly.
Post a picture of your device to check out the differences then.

andreasloe, 2018-08-24:
This is probably a misunderstanding: I am used to flash Sonoff and their pinouts are nicely labeled. I do not own a homecube yet but was wondering Howe to flash it - and looking and the above photographs I do not see any descriptions of pinouts.

closed 2018-08-26 by xoseperez

gretel, 2018-09-25:
@hyteoo 5V to `VCC` for flashing?

gretel, 2018-10-23:
`GPIO0` needs to be grounded to enable flash. all four LEDs will light cyan.
had to flash twice to get it working. seems to auto-reset when flashing is done.

janbernhardt, 2020-01-27:
This code seems to work pretty well with Gosund SP1 Switch. I was able to flash 3 devices without issues. Switch is working fine using button or with homeassistant (MQTT) integration.
One potential bug is that if I set the LED to "Always ON" the LED is off and if the set it to "Always OFF" the Button LED is on. So the behavior seems to be inverse. Also the power consumption looks to be wrong, but this is something I need to investigate more on.