xoseperez/espurna #1109: How secure is api communication

arihantdaga, 2018-08-06:
I am very new to web security. But I was wondering if the communication  using apiKey in espurna is very secure or not.. I guess if somebody sniffs the network, he can know the apiKey and then use that for future communication with the device.. Right ? 
But as i said i am very new to web security, but is there a way to secure this. I'll be happy to do a PR if somebody can brief me in theory what we need to do..

lobradov, 2018-08-07:
In general, anything with ESP8266 is *not* secure, due to one basic reason: due to memory constraints, there's practically no TLS implemented anywhere. This means that any interface (Web UI, MQTT, REST) is fully open and unencrypted, so all access keys are out on the open.

Having said that, it's a good practice to keep your device network separate from rest of the home network and at least in that way try to minimize attack vectors. This is far from "secure" but helps.

forast3r, 2018-08-08:
@lobradov Could you explain or give example of some good practices to increase security? how do you separate the networks with "soho" devices?

ServerCare, 2018-08-09:
you could implement VLAN's to seperate the sensors from your regular netwoprk, just make sure you have dns and ntp available for the espurna devices. also: once the esp32 is supported (by espurna, and used by sonoff) a lot more will be possble, such as bluetooth, TLS  and faster CPU.

mcspr, 2018-08-11:
btw recently released arduino core 2.4.2 now has bearssl as an alternative to the current axtls, which in some cases helps with memory issues and general stability when using tls client/server.
'Maximum Fragment Length Negotiation' setting being the main improvement, heap usage going down from current constant of 16kb buffer per connection to customizable 1kb/2kb/4kb etc. see bearssl pr discussion at Arduino/esp8266 for more info about that and [an example from the arduino repo](https://github.com/esp8266/Arduino/blob/e3c970210fe74d35049afd0a7dc06759d4f5c437/libraries/ESP8266WiFi/examples/BearSSL_MaxFragmentLength/BearSSL_MaxFragmentLength.ino)

Although, ESPurna defaults to the async libs that do not support it yet (espasynctcp being main one. espasyncwebserver and async-mqtt-client depending on it)

closed 2018-10-19 by stale[bot]