xoseperez/espurna #1110: Generic DHT

germaguire, 2018-08-06:
Hi, Could a Generic DHT build be added back into the prebuilt images, I have one from espurna-1.12.6a-generic-dht.bin which worked well on the Wemos D1 mini and Adafruit Huzzah boards but this doesn't seem to be available for the newer builds?

Thanks,
Ger.

germaguire, 2018-08-07:
I have tried to add this myself in the code but not sure if this is correct but it seems to work ok when I upload it. Is this correct ?

config/hardware.h

#elif defined(GENERIC_DHT)

    // Info
    #define MANUFACTURER        "GENERIC"
    #define DEVICE              "GENERIC_DHT"

    // DHT11
    #ifndef DHT_SUPPORT
    #define DHT_SUPPORT         1
    #endif
    #define DHT_PIN             14

platformio.ini

[env:generic-dht]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board_1m}
board_build.flash_mode = ${common.flash_mode}
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags_1m0m} -DGENERIC_DHT
monitor_speed = ${common.monitor_speed}
extra_scripts = ${common.extra_scripts}

xoseperez, 2018-08-13:
Yes, it's the basic configuration. It should work.

closed 2018-10-19 by stale[bot]