xoseperez/espurna #1115: MQTT on -> off freezes the device

jsponz, 2018-08-07:
With the last version, when switching off the MQTT, device freezes.  I have to plug it off and plug it on again. 

When switching on the MQTT, there is no problem. 

Any hint?

xoseperez, 2018-08-13:
Cannot reproduce the issue... can you check the debug log to see if there is any useful info there?

jsponz, 2018-08-13:
The problem is the library async-mqtt-client.

ESP just restarts and the web loses connection. You need to refresh the explorer.

xoseperez, 2018-08-14:
How did you get to that conclusion? What version/revision of the library are you using?

pniaps, 2018-08-17:
Hi @jsponz ,
When you hit `Save` button, if MQTT is enabled, it disconnects and connects again, and the problem is in the library async-mqtt-client. When it disconects, it crashes. The ESP reboots, but the web remains frozen until you reload it.

The problem is

https://github.com/marvinroger/async-mqtt-client/blob/33cdd5038b8b411db47322908ca4eb503f630909/src/AsyncMqttClient.cpp#L707

function **_sendDisconnect** closes connection on line 661, so **_client.send()** on line 707 crashes.

https://github.com/marvinroger/async-mqtt-client/blob/33cdd5038b8b411db47322908ca4eb503f630909/src/AsyncMqttClient.cpp#L661

I will send a pull request to https://github.com/marvinroger/async-mqtt-client

xoseperez, 2018-08-25:
Please, let me know when you send the PR.

Misiu, 2018-09-06:
@pniaps did You send PR for this?

Misiu, 2018-10-01:
@pniaps any news?

pniaps, 2018-10-14:
Hi.
pull request sent https://github.com/marvinroger/async-mqtt-client/pull/117

Sorry for the delay

mcspr, 2018-10-14:
@pniaps I'll wait for someone to answer the pr you proposed, but isn't that the case of having network request inside system callback? Inside WiFi event callback that is. If you change the callbacks to set boolean flags and do the connection inside loop() instead everything works ok. And, in espurna case, that is exactly what it does.

I did track this for some time but could not reproduce the original issue :/ 2.3.0 ... 2.4.2 did ok switching mqtt on / off / on again.

pniaps, 2018-10-14:
Will try, but callback or not, '_client.send();' is after '_client.close(true);'

closed 2018-12-25 by stale[bot]

mcspr, 2019-03-04:
Some notes, since PR was never merged and this is still a possible bug:
- reproducible only with async-mqtt-client git ((https://github.com/marvinroger/async-mqtt-client/commit/33cdd5038b8b411db47322908ca4eb503f630909), not point release version 0.8.1 
- lwip's tcp_output function, being called by the second _client->send(), crashes after _client->close() sets pcb to NULL.. But, only in system context. If called inside loop() it behaves just fine, because close() is not instant.
- esp32 async tcp library actually does check if pcb is not NULL, working around this bug: https://github.com/me-no-dev/AsyncTCP/pull/30/files