xoseperez/espurna #1117: relay 01 does not work

bruno-wladimir, 2018-08-07:
I have two modules t1 and the firmware change relay 01 does not work, even with a simple code where you activate and deactivate the relay, it does not work. relay 2 works perfectly in both the tasmota and a simple blink. Anyone have any idea what it can be?

xoseperez, 2018-08-13:
Do you mean a simple sketch does not work? Maybe the GPIO is wrong?

bruno-wladimir, 2018-08-13:
It worked before the same code, it suddenly stopped

xoseperez, 2018-08-13:
Maybe a hardware issue, can you test the relay isolated?

bruno-wladimir, 2018-08-15:
the relay will only work if you apply 5V to the power board pin

xoseperez, 2018-08-25:
That's because the relay is a 5V relay. The microcontroller uses 3V3 but cannot switch the relay by its own, it needs an external line rated for the relay coil.

closed 2018-10-31 by stale[bot]