xoseperez/espurna #1119: espurna-1.13.1-blitzwolf-bwshp2.bin fairst when changing admin password

kroon040, 2018-08-08:
Hi,

I bought an blitzwolf bwshp2 , and i flashed it with the 
https://github.com/xoseperez/espurna/releases/download/1.13.1/espurna-1.13.1-blitzwolf-bwshp2.bin

After connecting to the ssid and I also get the web interface, but when changing the password, I thnink the the device reboots, and I can't connect to the device anymore (not to the ssid), when flashing again, the same happens.

Any idea , I took Admin123 as password

regards Richard

flashy02, 2018-08-10:
Hi,
I had the same behaviour with my today flashed "Homecube 16A". Every time I tried to change the PWD, the device reboots. I'have tried several times with 230V connected (Plug), or 3,3V connected (serial flasher). And the behaviour was nearly the same. After a lot of tries, I got a step further, and the device stored the new "pwd". Another more tries/reboots, the devices stored the configuration and some more tries I got it to connect to my wifi. After these round about 30 tries/reboots my device is configurated and it works. But this "first" installation is not really useable.
I have to say it is my first device wich I haved flashed out of the box to 1.13.1 master. So I have no comparison if it is a problem of the device or the 1.13.1.
All my updated devices from 1.12.3-1.12.6 to 1.13.1 (last week) were without problems.

Regards Michael

flashy02, 2018-08-10:
One point I forgot. After changing the PWD (with 3,3v power) I wasn't able to connect while connected to 230V (not fibonacci and not with new pwd). With the serial programmer 3.3V it was possible. Now after I finished the configuration it works with 3,3v and 230v. I don't understand this behaviour.

kroon040, 2018-08-11:
Hi,

I solved it , changing the admin password, also changed the wpa2 password to the same.

So I removed the network and reconnected it with the wpa2 password (also the admin password)

regards Richard

xoseperez, 2018-08-13:
Yes, the admin password is used for pretty much any "admin" access (web interface, wifi, OTA,...). So remember to use the new password for the new wifi AP.

closed 2018-08-13 by xoseperez

kakaki, 2018-08-28:
A have similar issue with 1.13.1 and blitzwolf - device flashed with fixed wifi settings - it's working, updating default password works, but saving settings with simple change as led status reboots device and it goes back to default with default password.

ruimarinho, 2018-09-08:
I've had a similar issue with a Sonoff POW R2 too.