xoseperez/espurna #1129: Question - Multiple Digital Sensors

saundersrg, 2018-08-12:
Hi there, perhaps a silly question, but how can I add multiple digital sensors? I see that relays, switches, led's etc are numbered, but other sensors are not. I have a project that has 2x radar units (binary sensor) connected to 2x pins, how can I enable them both?

Thanks!

xoseperez, 2018-08-13:
Magnitudes (not sensors) are automatically numbered but at the moment there is no way to specify the numbering you want. So if you have two radar sensors messages will go to `<root>/distance/0` and `<root>/distance/1`.

Now, to load both, and until there is no soft-config way to do it, you will have to do it in code, editing the sensors.ino file. Please check the `_sensorLoad` method where you can find an example on how to do it (with a DHT sensor, just adapt it to your use case).

saundersrg, 2018-08-14:
Excellent, thank you I'll try it out!

closed 2018-08-14 by saundersrg