xoseperez/espurna #1130: ESP-01 + 1CH 5V RELAY FROM LC TECHNOLOGY

rossmax, 2018-08-12:
Hi!

I try to modificate some parts of code to make it work. I have ESP-01 + 1CH 5V RELAY FROM LC TECHNOLOGY, https://www.az-delivery.de/products/esp8266-01s-mit-relais?ls=de , https://www.hackster.io/makerrelay/esp8266-wifi-5v-1-channel-relay-delay-module-iot-smart-home-e8a437 

This board does not use GPIO0 or GPIO2 in order to activate the relay, it use the serial communication instead it.
First: you need to open Serial port communication at 9600B, any other speed does not works fine.
Ex. : Serial.begin(9600);

Second: 
To enable the Relay send it by serial port:

const byte miBufferON[] = {0xA0, 0x01, 0x01, 0xA2};
Serial.write(miBufferON, sizeof(miBufferON));

To disable the Relay send it by serial port:
const byte miBufferOFF[] = {0xA0, 0x01, 0x00, 0xA1};
Serial.write(miBufferON, sizeof(miBufferOFF));

I add in hardware.h

```
#elif defined(GENERIC_ESP01S_LCTECH_RELAY)
    // Remove UART noise on serial line
    #define DEBUG_SERIAL_SUPPORT    0

    #define MANUFACTURER        "GENERIC"
    #define DEVICE              "LCTECH_RELAY"
    #define RELAY_PROVIDER          RELAY_PROVIDER_LCTECH
    #define SERIAL_BAUDRATE         9600

    // Relays
    #define DUMMY_RELAY_COUNT             1

    // Buttons
    #define BUTTON1_PIN         	       3    //external button connected to "Rx" pin
    #define BUTTON1_MODE        	       BUTTON_PUSHBUTTON
    #define BUTTON1_PRESS       	       BUTTON_MODE_TOGGLE
    #define BUTTON1_CLICK       	       BUTTON_MODE_NONE
    #define BUTTON1_DBLCLICK    	       BUTTON_MODE_NONE
    #define BUTTON1_LNGCLICK    	       BUTTON_MODE_NONE
    #define BUTTON1_LNGLNGCLICK 	       BUTTON_MODE_NONE
    #define BUTTON1_RELAY       	        1

    // LEDs
    #define LED1_PIN                 1
    #define LED1_MODE                LED_MODE_RELAY
    #define LED1_PIN_INVERSE         1
    #define LED1_RELAY               1

#endif
```

in platfomio.ini 
```
[env:generic-esp01s-lctech-relay]
platform = ${common.platform}
framework = arduino
board = esp01
board_flash_mode = dout
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags_512k} -DGENERIC_ESP01S_LCTECH_RELAY
monitor_baud = 115200
extra_scripts = ${common.extra_scripts}
```

in relay.ino 

```
 #if RELAY_PROVIDER == RELAY_PROVIDER_LCTECH
             pinMode(LED1_PIN, INPUT);    // need in order to get back from external led mode to serial
             delay(5);
            Serial.begin(9600);   //9600 for this module and avoid any other serial usage 
            const byte ON[] = {0xA0,0x01,0x01,0xA2};
            Serial.write(ON,sizeof(ON));
            Serial.flush();
           Serial.end();    //closing serial since we need to go to led digital pin output
            pinMode(LED1_PIN, OUTPUT);    //for led manager 
       #endif
```
here i just try to put relay in on mode, but unfortunately its not working.  Simple arduino sketch work fine, but  i want to make it work with espurna. 

Here is simple sketch 

```
byte relON[] = {0xA0, 0x01, 0x01, 0xA2};  //Hex command to send to serial for open relay
byte relOFF[] = {0xA0, 0x01, 0x00, 0xA1}; //Hex command to send to serial for close relay

int ledState = false;
unsigned long previousMillis = 0;
const long interval = 2000; //  2 seconds

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize serial:
  Serial.begin(9600);
}

// the loop function runs over and over again forever
void loop()
{
  unsigned long currentMillis = millis();
  if(currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;   
    if (ledState == true) {
      Serial.write(relON, sizeof(relON));     // turns the relay ON
      ledState = false;
    } else {
      Serial.write(relOFF, sizeof(relOFF));   // turns the relay OFF
      ledState = true;
    }    
  }
}
```

rossmax, 2018-08-12:
I found solution, i dont know why i must send hex command 4 times, but it works.

```
#if RELAY_PROVIDER == RELAY_PROVIDER_LCTECH
             pinMode(LED1_PIN, INPUT);    // need in order to get back from external led mode to serial
             delay(5);
            Serial.begin(9600);   //9600 for this module and avoid any other serial usage 
            byte relON[] = {0xA0, 0x01, 0x01, 0xA2};  //Hex command to send to serial for open relay
            byte relOFF[] = {0xA0, 0x01, 0x00, 0xA1}; //Hex command to send to serial for close relay
            if (status == 0x01) {
                Serial.write(relON,sizeof(relON));
                Serial.write(relON,sizeof(relON));
                Serial.write(relON,sizeof(relON));
                Serial.write(relON,sizeof(relON));
            }
            if (status == 0x00) {
                Serial.write(relOFF,sizeof(relOFF));
                Serial.write(relOFF,sizeof(relOFF));
                Serial.write(relOFF,sizeof(relOFF));
                Serial.write(relOFF,sizeof(relOFF));
            }
            Serial.flush();
           Serial.end();    //closing serial since we need to go to led digital pin output
            pinMode(LED1_PIN, OUTPUT);    //for led manager 
       #endif
```

mcspr, 2018-08-12:
looking at relay.ino providers i noticed RELAY_PROVIDER_STM which uses exact same protocol. have you tried it?
https://github.com/xoseperez/espurna/blob/4fa1c9d1e076ef7100f7d2e14afb178d98893088/code/espurna/relay.ino#L80-L87

xoseperez, 2018-08-13:
The RELAY_PROVIDER_STM protocol is exactly the same, it should work. But I have noticed the provider does not properly configure the baud rate, maybe adding `Serial.begin(9600)` before the first flush would do...
If it worked after sending it multiple times it looks like serial noise to me. Doing a flush like the RELAY_PROVIDER_STM does might do the trick.

rossmax, 2018-08-14:
xoseperez  100% right! Serial.begin(9600) and Serial.flush() resolve problem. Then
 in relay.ino 
```
#if RELAY_PROVIDER == RELAY_PROVIDER_LCTECH
            Serial.flush();
            Serial.begin(SERIAL_BAUDRATE);
            Serial.write(0xA0);
            Serial.write(id + 1);
            Serial.write(status);
            Serial.write(0xA1 + status + id);
            Serial.flush();
       #endif
```
```
#if RELAY_PROVIDER == RELAY_PROVIDER_STM || RELAY_PROVIDER == RELAY_PROVIDER_LCTECH
            _relays[i].change_time = millis() + 3000 + 1000 * i;
        #else
```
and hardware.h
```
#elif defined(GENERIC_ESP01S_LCTECH_RELAY)
    // Remove UART noise on serial line
    #define DEBUG_SERIAL_SUPPORT    0

    #define MANUFACTURER        "GENERIC"
    #define DEVICE              "LCTECH_RELAY"
    #define RELAY_PROVIDER          RELAY_PROVIDER_LCTECH
    #define SERIAL_BAUDRATE         9600

    // Relays
    #define DUMMY_RELAY_COUNT       1

    // LEDs
    #define LED1_PIN                 1
    #define LED1_MODE                LED_MODE_RELAY
    #define LED1_PIN_INVERSE         1
    #define LED1_RELAY               1

#endif
```

closed 2018-08-14 by rossmax

mcspr, 2018-08-14:
Great.

btw you do specify SERIAL_BAUDRATE which would've called this part, but never does (only checks for RELAY_STM device and not just RELAY_PROVIDER_STM): https://github.com/xoseperez/espurna/blob/373fc535452a5e91b8fb81954d58c471eede3724/code/espurna/system.ino#L143-L145

xoseperez, 2018-08-14:
@mcspr You are right. I changed the check. BTW, @rossmax, you might want to simply use RELAY_PROVIDER_STM in you board definition instead of creating a new relay provider.

dezral, 2018-12-15:
i have one of those with only (LC v3) 1 channel, and have one with (link)4 channels on the way in the mail.. but i can't get the 1channel to work?  either channel it shows don't work..

[https://www.banggood.com/DC5V-ESP8266-Four-Channel-Wifi-Relay-IOT-Smart-Home-Phone-APP-Remote-Control-Switch-p-1317109.html?utm_source=adwords&utm_campaign=ios-share-ds-elc-cs-mkt-bg-url_m_product&utm_medium=cpc_elc&utm_content=serene&utm_design=0&ios_share=1&_branch_match_id=453122632230721132](url)

Entepotenz, 2019-02-22:
> stm-relay
> i have one of those with only (LC v3) 1 channel, and have one with ( [Link](https://www.banggood.com/DC5V-ESP8266-Four-Channel-Wifi-Relay-IOT-Smart-Home-Phone-APP-Remote-Control-Switch-p-1317109.html?utm_source=adwords&utm_campaign=ios-share-ds-elc-cs-mkt-bg-url_m_product&utm_medium=cpc_elc&utm_content=serene&utm_design=0&ios_share=1&_branch_match_id=453122632230721132) )4 channels on the way in the mail.. but i can't get the 1channel to work? either channel it shows don't work..
> 
> okay it seems my LC relay v3 does not have a smt chip but n76e003at20 microprocessor.. don't know where to go from now and debug..
> 
> this the modules exactly i have:
> [link](https://www.aliexpress.com/item/Free-Shipping-ESP8266-5V-Wifi-Relay-Module-Remote-Control-Switch-Phone-APP-For-Smart-Home-IOT/32789132806.html?spm=2114.search0104.3.103.22942a144P4FgV&ws_ab_test=searchweb0_0,searchweb201602_3_10065_10068_319_317_10696_10084_453_454_10083_10618_10304_10307_10820_10821_538_537_10302_536_10059_10884_10887_100031_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=6b58425f-4416-4f49-9c82-1e393f98101c-14&algo_pvid=6b58425f-4416-4f49-9c82-1e393f98101c)
> 
> link for more details:
> https://www.behind-the-scenes.co.za/esp8266-5v-wi-fi-relay-switch-modules/

Hey i have the same problem as you. (my 2 channel relay uses the **Nuvoton N76E003AT20**)
I did some debugging and was able to read the serial tx of the esp8266 while booting and switching the relays on/off.
I used minicom with an **Baudrate of 115200** with 8N1.
I attached the full Serial printout on tx while booting to this comment. It contains the booting process and turning on/off both relays.
[minicom.zip](https://github.com/xoseperez/espurna/files/2893631/minicom.zip)

On the website of lctech you can see some information about the 2 channel relay chip (the website does not seem to work properly at the moment therefore i provide a link to archive.org): [https://web.archive.org/web/20180602202511/http://www.chinalctech.com/index.php?_m=mod_product&_a=view&p_id=1261](https://web.archive.org/web/20180602202511/http://www.chinalctech.com/index.php?_m=mod_product&_a=view&p_id=1261)

it writes that you need those hexcodes to switch the relays:
Open the first relay：   ``A0 01 01 A2``
Close the first Relay：  ``A0 01 00 A1``
Open the second relay： ``A0 02 01 A3``
Close the second Relay： ``A0 02 00 A2 ``

an ``hexdump -C`` of the file **minicom_relay1_open.cap** shows this:
``00000000  0a 2b 49 50 44 2c 30 2c  34 3a a0 01 01 a2        |.+IPD,0,4:....|``
an ``hexdump -C`` of the file **minicom_relay1_close.cap** shows this:
``00000000  0a 2b 49 50 44 2c 30 2c  34 3a a0 01 a1           |.+IPD,0,4:...|``
an ``hexdump -C`` of the file **minicom_relay2_open.cap** shows this:
``00000000  0a 2b 49 50 44 2c 30 2c  34 3a a0 02 01 a3        |.+IPD,0,4:....|``
an ``hexdump -C`` of the file **minicom_relay2_close.cap** shows this:
``00000000  0a 2b 49 50 44 2c 30 2c  34 3a a0 02 a2           |.+IPD,0,4:...|``

For unknown reason the ``\x00`` is omitted in close first and second relay. This could be caused by minicom? ``¯\_(ツ)_/¯``

However, you can see the hexcodes provided by lctech.
I think the precedent code ``0a 2b 49 50 44 2c 30 2c  34 3a`` is the required configuration of the nuvoton.
I have not had time to setup the esp8266 and test it for myself yet. But maybe this helps someone :)

dezral, 2019-02-22:
I just use tasmato now with it and some serialsend5 command.. and some “dummy” relays.. and then a rule that listens for switching and then a rule that sendes the command..

he-leon, 2019-03-19:
Hi! 
I'm also trying to get the Nuvoton Version of this relay to work. 
@Entepotenz, @dezral : I tried to simulate the "handshake" of the ESP and the Nuvoton based on your serial log, but I didn't succeed. Unfortunately I didn't make a backup of the original firmware of the ESP01 before flashing Tasmota, so I cannot restore the original firmware to check the serial communication.
Could you maybe upload a backup of the original firmware? That would be great!
This should dump the 1MB (or is it a 512KB version?) flash of the ESP01 into backup.bin:
`esptool.py --port /dev/ttyUSB0 read_flash 0x00000 0x100000 ./backup.bin`
Thank you!
PS:  I'm not entirely sure but I think the EEPROM is also emulated in flash, so you should probably reset the WiFi credentials before posting a backup.

he-leon, 2019-03-25:
Hi,
i got my 4 channel Nuvoton based relay working.
I added a nuvoton-4ch-relay env (see diff). Hope this helps someone. I made the changes in version 1.13.3, because the webinterface didn't work for me in 1.13.5 (blank page).

```
diff --git a/code/espurna/config/hardware.h b/code/espurna/config/hardware.h
index 6068a274..a02570b2 100644
--- a/code/espurna/config/hardware.h
+++ b/code/espurna/config/hardware.h
@@ -1806,6 +1806,23 @@
     // Remove UART noise on serial line
     #define DEBUG_SERIAL_SUPPORT    0
 
+// -----------------------------------------------------------------------------
+// NUVOTON 4CH RELAY
+// -----------------------------------------------------------------------------
+
+#elif defined(NUVOTON_4CH_RELAY)
+
+    // Info
+    #define MANUFACTURER            "LCTech"
+    #define DEVICE                  "4CH Relay"
+
+    // Relays
+    #define DUMMY_RELAY_COUNT       4
+    #define RELAY_PROVIDER          RELAY_PROVIDER_NUVOTON
+
+    // Remove UART noise on serial line
+    #define DEBUG_SERIAL_SUPPORT    0
+
 // -----------------------------------------------------------------------------
 // Tonbux Powerstrip02
 // -----------------------------------------------------------------------------
diff --git a/code/espurna/relay.ino b/code/espurna/relay.ino
index df01fbc4..66281ff0 100644
--- a/code/espurna/relay.ino
+++ b/code/espurna/relay.ino
@@ -86,6 +86,25 @@ void _relayProviderStatus(unsigned char id, bool status) {
         Serial.flush();
     #endif
 
+    #if RELAY_PROVIDER == RELAY_PROVIDER_NUVOTON
+        Serial.flush();
+        Serial.write(0x0a);
+        Serial.write(0x2b);
+        Serial.write(0x49);
+        Serial.write(0x50);
+        Serial.write(0x44);
+        Serial.write(0x2c);
+        Serial.write(0x30);
+        Serial.write(0x2c);
+        Serial.write(0x34);
+        Serial.write(0x3a);
+        Serial.write(0xA0);
+        Serial.write(id + 1);
+        Serial.write(status);
+        Serial.write(0xA1 + status + id);
+        Serial.flush();
+    #endif
+
     #if RELAY_PROVIDER == RELAY_PROVIDER_LIGHT
 
         // If the number of relays matches the number of light channels
diff --git a/code/espurna/system.ino b/code/espurna/system.ino
index c988c6cd..1cd75e57 100644
--- a/code/espurna/system.ino
+++ b/code/espurna/system.ino
@@ -144,7 +144,7 @@ void _systemSetupSpecificHardware() {
 
     // These devices use the hardware UART
     // to communicate to secondary microcontrollers
-    #if defined(ITEAD_SONOFF_RFBRIDGE) || defined(ITEAD_SONOFF_DUAL) || (RELAY_PROVIDER == RELAY_PROVIDER_STM)
+    #if defined(ITEAD_SONOFF_RFBRIDGE) || defined(ITEAD_SONOFF_DUAL) || (RELAY_PROVIDER == RELAY_PROVIDER_STM || RELAY_PROVIDER == RELAY_PROVIDER_NUVOTON)
         Serial.begin(SERIAL_BAUDRATE);
     #endif
 
diff --git a/code/platformio.ini b/code/platformio.ini
index 9d81cf13..692357dc 100644
--- a/code/platformio.ini
+++ b/code/platformio.ini
@@ -72,7 +72,7 @@ extra_scripts = extra_scripts.py
 #   Please note that we don't always use the latest version of a library.
 # ------------------------------------------------------------------------------
 lib_deps =
-    ArduinoJson
+    https://github.com/bblanchon/ArduinoJson#v5.13.5
     https://github.com/marvinroger/async-mqtt-client#v0.8.1
     Brzo I2C
     https://github.com/xoseperez/debounceevent.git#2.0.4
@@ -1956,12 +1956,23 @@ board_build.flash_mode = ${common.flash_mode}
 lib_deps = ${common.lib_deps}
 lib_ignore = ${common.lib_ignore}
 build_flags = ${common.build_flags_1m0m} -DSTM_RELAY
-upload_speed = ${common.upload_speed}
 upload_port = ${common.upload_port}
 upload_flags = ${common.upload_flags}
 monitor_speed = ${common.monitor_speed}
 extra_scripts = ${common.extra_scripts}
 
+[env:nuvoton-4ch-relay]
+platform = ${common.platform}
+framework = ${common.framework}
+board = ${common.board_1m}
+board_build.flash_mode = ${common.flash_mode}
+lib_deps = ${common.lib_deps}
+lib_ignore = ${common.lib_ignore}
+upload_speed = ${common.upload_speed_fast}
+build_flags = ${common.build_flags_1m0m} -DNUVOTON_4CH_RELAY
+monitor_speed = ${common.monitor_speed}
+extra_scripts = ${common.extra_scripts}
+
 [env:heygo-hy02]
 platform = ${common.platform}
 framework = ${common.framework}
```

megamarco833, 2019-03-25:
hi, i have relay 2ch with Nuvoton N76: n76e003at20
what i should do to make it working?
i have to flash the chip n76e003at20 ? with what tools and firmware?
thanks!!

hi, i have relay 2ch with Nuvoton N76: n76e003at20
what i should do to make it working?
i have to flash the chip n76e003at20 ? with what tools and firmware?
thanks!!

megamarco833, 2019-03-25:
> 
> 
> Hi,
> I got my 4 channel Nuvoton based relay working.
> I added a nuvoton-4ch-relay env (see diff). Hope this helps someone. I made the changes in version 1.13.3, because the webinterface didn't work for me in 1.13.5 (blank page).
> 
hi, i have relay 2ch with Nuvoton N76: n76e003at20
what i should do to make it working?
i have to flash the chip n76e003at20 ? with what tools and firmware?
thanks!!

he-leon, 2019-03-25:
You have to build espurna yourself using [PatformIO](https://github.com/xoseperez/espurna/wiki/PlatformIO). Before building you have to include the changes i made from the diff above in the following files: 
```
code/espurna/config/hardware.h
code/espurna/relay.ino
code/espurna/system.ino
code/platformio.ini
```

megamarco833, 2019-03-25:
i never succeed on using platformIO to compile espurna :(
i also tried with arduino ide
could you please attach your bin fine?
thanks!

mcspr, 2019-03-26:
@he-leon see https://github.com/xoseperez/espurna/issues/1610#issuecomment-473794750 about the webui problem. You can check the 1.13.5 with those changes

he-leon, 2019-03-26:
@mcspr Thanks for the hint. 

> i never succeed on using platformIO to compile espurna :(
> i also tried with arduino ide
> could you please attach your bin fine?
> thanks!

@megamarco833, @xoseperez I'm not sure if it's okay for the maintainers to upload unofficial binaries here. Generally it's not a good idea to run binaries from random people on a device in your private network.

karmacoma92, 2019-04-09:
hi he-leon
I've tried your solution adapted to a 2ch relay and it partly works, I can switch both relays ON and OFF but when I try to switch both ON (one after the other) independently on which one I switch ON first, both go OFF...
any ideas?
thanks in advance

btw, I did an original sw backup with the command you mention above but I cannot restore it now, it keeps on giving me an error command after the flash is completed, something about the header....

he-leon, 2019-04-10:
hi, 
I have also experienced problems when switching different relays very quickly (maybe < 240ms) after each other. I think the nuvoton get's confused when multiple commands are sent in a short time. However, I didn't do any further research on this problem, because for my use case it works fine.

Hm, I don't really know what went wrong with your backup. Maybe you have a 512K version of the esp-01. The commands i posted should work for the 1M version. I'm not sure what would happen with a 512K version.

karmacoma92, 2019-04-10:
hi he-leon, 
thanks for your answer, it doesn't matter how much time I take to switch the second relay after switching the first one, they do always turn off, but I'll try to play around a little...

Regarding the backup I made sure it was a 1M version with command "esptool.py --port /dev/ttyUSB0 flash_id", do not know what happened because I used the very same command you posted...
thanks again for your help!!

mcspr, 2019-04-10:
@he-leon If you do not mind, I'll add the patch you provided to the dev tree.
When someone figures out a fix for the interlocking problem, they can send a PR.

Entepotenz, 2019-04-10:
I am neither an esp8266 expert nor a nuvoton expert and i did not try switching the relays faster than 240ms but maybe interrupts could help.

This code runs fine:
```C
noInterrupts();
Serial.flush();
Serial.begin(SERIAL_BAUDRATE);
Serial.write(...); // send commands
Serial.write('\n');
Serial.flush();
interrupts();`
```

mcspr, 2019-04-10:
Maybe this is the reason?
```
Serial.write('\n');
```
Bytes from https://github.com/xoseperez/espurna/issues/1130#issuecomment-476383929 above:
```python
>>> bytes([int(x, 16) for x in ['0xa', '0x2b', '0x49', '0x50', '0x44', '0x2c', '0x30', '0x2c', '0x34', '0x3a', '0x01', '0x01', '0xa3']])
b'\n+IPD,0,4:\x01\x01\xa3'
```
There's newline, but it is at the start. Maybe device thinks that is an empty command?

karmacoma92, 2019-04-11:
Maybe, since I tried what Enteponez suggested above and now it works with both the STM approach and the he-leon one, ....

he-leon, 2019-04-11:
@mcspr Sure you can add the patch. 
Regarding the byte sequence: I just used the sequence from @Entepotenz Serial dump. I didn't give it any further thought, since it worked for me. But yes, The 0xa newline at the beginning seems a bit odd. It may be worth a try to add the newline at the end.
`b'+IPD,0,4:\x01\x01\xa3\n'`

Apparently the different boards behave slightly different, because my relay works fine with my patch. Maybe we can find a solution working for all versions.

karmacoma92, 2019-04-11:
Hi there, I tested the \n solution without the interrupts and it works for me. I have the impression that once the NUVOTON works once it does always work because it is working with both the STM and the NUVOTON approach
the NUVOTON approach :

> 
>     #if RELAY_PROVIDER == RELAY_PROVIDER_NUVOTON
>         //noInterrupts();
>         Serial.flush();
>         Serial.begin(SERIAL_BAUDRATE);
>         Serial.write(0x0a);
>         Serial.write(0x2b);
>         Serial.write(0x49);
>         Serial.write(0x50);
>         Serial.write(0x44);
>         Serial.write(0x2c);
>         Serial.write(0x30);
>         Serial.write(0x2c);
>         Serial.write(0x34);
>         Serial.write(0x3a);
>         Serial.write(0xA0);
>         Serial.write(id + 1);
>         Serial.write(status);
>         Serial.write(0xA1 + status + id);
>         Serial.write('\n');
>         delay(100);
>         Serial.flush();
>         //interrupts();
>     #endif

and the STM approach

> 
>     #if RELAY_PROVIDER == RELAY_PROVIDER_STM
>         Serial.flush();
>         Serial.write(0xA0);
>         Serial.write(id + 1);
>         Serial.write(status);
>         Serial.write(0xA1 + status + id);
> 
>         // The serial init are not full recognized by relais board.
>         // References: https://github.com/xoseperez/espurna/issues/1519 , https://github.com/xoseperez/espurna/issues/1130
>         Serial.write('\n');
>         delay(100);
> 
>         Serial.flush();
>     #endif

it not clear to me whether this solution would work in every version of the board or not, this both are working now for me....

Regards

karmacoma92, 2019-04-11:
Hi there, I tested the \n solution without the interrupts and it works for me. I have the impression that once the NUVOTON works once it does always work because it is working with both the STM and the NUVOTON approach
the NUVOTON approach :

> #if RELAY_PROVIDER == RELAY_PROVIDER_NUVOTON
>         //noInterrupts();
>         Serial.flush();
>         Serial.begin(SERIAL_BAUDRATE);
>         Serial.write(0x0a);
>         Serial.write(0x2b); [...]

and the STM approach

> #if RELAY_PROVIDER == RELAY_PROVIDER_STM
>         Serial.flush();
>         Serial.write(0xA0);
>         Serial.write(id + 1);
>         Serial.write(status);
>         Serial.write(0xA1 + status + [...]

it not clear to me whether this solution would work in every version of the board or not, this both are working now for me....

Regards

he-leon, 2019-04-11:
Hi,
turns out my relay works with the standard STM commands. :man_facepalming:  I have no Idea why I couldn't get it to work in the first place. That's why I implemented the command from @Entepotenz Serial dump. Sorry, for the confusion!
So I guess, @mcspr, there's no need to add my patch to the dev branch. The only changes to get my relay (LCTech 4CH Nuvoton based) relay working are the following:

**hardware.h**
```
// -----------------------------------------------------------------------------
// LCTech 4CH relay with Nuvoton 
// -----------------------------------------------------------------------------

#elif defined(NUVOTON_4CH_RELAY)

    // Info
    #define MANUFACTURER            "LCTech"
    #define DEVICE                  "4CH Relay"

    // Relays
    #define DUMMY_RELAY_COUNT       4
    #define RELAY_PROVIDER          RELAY_PROVIDER_STM

    // Remove UART noise on serial line
    #define DEBUG_SERIAL_SUPPORT    0
```
**platformio.ini**
```
[env:nuvoton-4ch-relay]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board_1m}
board_build.flash_mode = ${common.flash_mode}
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
upload_speed = ${common.upload_speed_fast}
build_flags = ${common.build_flags_1m0m} -DNUVOTON_4CH_RELAY
monitor_speed = ${common.monitor_speed}
extra_scripts = ${common.extra_scripts}

[env:nuvoton-4ch-relay-ota]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board_1m}
board_build.flash_mode = ${common.flash_mode}
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags_1m0m} -DNUVOTON_4CH_RELAY
upload_speed = ${common.upload_speed}
upload_port = ${common.upload_port}
upload_flags = ${common.upload_flags}
extra_scripts = ${common.extra_scripts}
monitor_speed = ${common.monitor_speed}
```

Here are the working HEX sequences for my LCTech 4CH Nuvoton based Relay
```
First relay: 
ON: a0 01 01 a2 
OFF: a0 01 00 a1 

Second relay: 
ON: a0 02 01 a3 
OFF: a0 02 00 a2 

Third relay: 
ON: a0 03 01 a4  
OFF: a0 03 00 a3  

Fourth relay: 
ON: a0 04 01 a5 
OFF: a0 04 00 a4 
```

santeceletrica, 2019-09-25:
Arduino Code:

Drive of the 04 relays running in sequence. time

///https://www.banggood.com/DC12V-ESP8266-Four-Channel-Wifi-Relay-IOT-Smart-Home-Phone-APP-Remote-Control-Switch-p-1317255.html?rmmds=myorder&cur_warehouse=CN

#define CH_PD 44 //sinal de controle de CH_PD
#define RST 46 //sinal de controle de RST
#define GPIO0 48 //sinal de controle de GPIO0
uint8_t R3On[]  = {0xA0, 0x01, 0x01, 0xA2};
uint8_t R3Off[] = {0xA0, 0x01, 0x00, 0xA1};
uint8_t R4On[]  = {0xA0, 0x02, 0x01, 0xA3};
uint8_t R4Off[] = {0xA0, 0x02, 0x00, 0xA2};
uint8_t R1On[]  = {0xA0, 0x03, 0x01, 0xA4};
uint8_t R1Off[] = {0xA0, 0x03, 0x00, 0xA3};
uint8_t R2On[]  = {0xA0, 0x04, 0x01, 0xA5};
uint8_t R2Off[] = {0xA0, 0x04, 0x00, 0xA4};
void setup() {
  pinMode(CH_PD,OUTPUT);
  pinMode(RST,OUTPUT);
  pinMode(GPIO0,OUTPUT);
  digitalWrite(CH_PD,HIGH); //Setado em alto - funcionamento normal
  digitalWrite(RST,HIGH); //RST em alto - funcionamento normal
  digitalWrite(GPIO0,HIGH); //GPIO0 em alto - funcionamento no rmal
  // Inicializa ambas as portas

  Serial.begin(115200);

}

void loop() {

    
    Serial.write(R1On, 4);
 delay(1000);
    Serial.write(R1Off, 4);
  delay(1000);
      Serial.write(R2On, 4);
 delay(1000);
    Serial.write(R2Off, 4);
  delay(1000);
      Serial.write(R3On, 4);
 delay(1000);
    Serial.write(R3Off, 4);
  delay(1000);
      Serial.write(R4On, 4);
 delay(1000);
    Serial.write(R4Off, 4);
  delay(1000);
}

marcorighi, 2021-11-03:
Hi,
I have a problem. Often changing the state of the relay (I have a board with only one relay) the ESP is reset... and I found the variables to the initial state. Quickly, I have the variable isStatusOk set to z (zero) when the board starts and I set the variable isStatusOk to o (one) with a command. I use this strategy to track power loss in my house. This ESP with the board has another task to execute, switch on and set off a light. 
Turning the light on/off the ESP is reset... the Relay drives the 220V and works as a switch... Follows the code.
I tried various strategies... now the board with relay and ESP has a 10 millifarad capacitor In input and my tester does not go under 5.2 volts... Any ideas?

`#include <ESP8266WiFi.h>

#define LED1_PIN 1
 
const char* ssid = "wlan-garden";
const char* password = "fraumb2016AAAABBBB";
 
WiFiServer server(2323);
char isStatusOk='z';

const byte relNO[] = {0xA0, 0x01, 0x01, 0xA2};  //Hex command to send to serial for open relay
const byte relNC[] = {0xA0, 0x01, 0x00, 0xA1}; //Hex command to send to serial for close relay
 
void setup() {

  // Connect to WiFi network
 
  WiFi.begin(ssid, password);
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    // Serial.print(".");
  }

 
  // Start the server
  server.begin();
  isStatusOk='z';
 
}
 
void loop() {

  char request;
  char symbolToSend;
    
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    return;
  }

  symbolToSend='x';
 
  // Wait until the client sends some data
  while(!client.available()){
    delay(1);
  }

  //// Read the first line of the request
  while (client.available()){
    request = client.read();
    // Serial.println(request);
    //client.flush();
    //Serial.flush();
    // Match the request
   
    //int value = LOW;
    if (request == 'z') {
      isStatusOk='z';
      symbolToSend=isStatusOk;    
    } 
    if (request == 'o') {
      isStatusOk='o';
      symbolToSend=isStatusOk;
    } 

    if ( (request == 'a') || (request == 's')) {
      pinMode(LED1_PIN, INPUT);    // need in order to get back from external led mode to serial
      delay(10);
      Serial.begin(9600);
      delay(10);
      Serial.flush();
      delay(20);
    }

    //s pegni dispositivo esterno
    if (request == 'a') {
      symbolToSend='a';
      Serial.write(relNC,sizeof(relNC));
      
    } 
    //a ccendi dispositivo esterno
    if (request == 's') {
      symbolToSend='s';
      Serial.write(relNO,sizeof(relNO));
    }     

    if ( (request == 'a') || (request == 's')) {
      delay(20);
      Serial.flush();
      delay(100);
      Serial.end();    //closing serial since we need to go to led digital pin output
      delay(20);
      pinMode(LED1_PIN, OUTPUT);    //for led manager 
    }    
  
    if (request == 'v' ) {
      symbolToSend=isStatusOk;
    } 

    client.write(symbolToSend);
    client.write('\n');
    client.flush();
    delay(100);
    client.stop();
  }

}`

Thx

linux01d, 2021-11-22:
> Hi! I'm also trying to get the Nuvoton Version of this relay to work. @Entepotenz, @dezral : I tried to simulate the "handshake" of the ESP and the Nuvoton based on your serial log, but I didn't succeed. Unfortunately I didn't make a backup of the original firmware before flashing, so I cannot restore the original firmware to check the serial communication. Could you maybe upload a backup of the original firmware? That would be great! This should dump the 1MB (or is it a 512KB version?) flash of the ESP01 into backup.bin: `esptool.py --port /dev/ttyUSB0 read_flash 0x00000 0x100000 ./backup.bin` Thank you! PS: I'm not entirely sure but I think the EEPROM is also emulated in flash, so you should probably reset the WiFi credentials before posting a backup.

May be somebody find it useful. I think it is better too late, than never in this case.
Original firmware for LC Tech ESP01 4R 12V is [here](https://disk.yandex.ru/d/e4P3L7eP1kFQRA). This dump I've made from ESP-01, which supplied with relay board based on nuvoton ms51fb9ae.