xoseperez/espurna #1133: Add more than 8 switches in Sonoff RF Bridge

vaidyasr, 2018-08-14:
I would like to add more than 8 switches from the default prebuilt binary. I have modified the arduino.h and hardware.h to accommodate 16 switches and successfully uploaded to the RF Bridge. After the web page loads, the page is randomly not loaded properly and the RF switch capturing page is not capturing full codes of RF. What am I missing?

xoseperez, 2018-08-14:
Apparently, increasing the number of switches have a big impact on available memory. I'm trying to find the reason.

xoseperez, 2018-08-14:
OK, api module was the culprit here, it was taking a lot of memory for each extra relay. You can checkout dev branch and update the hardware.h file to support 16 relays without issue. From them on other memory issues arise with the web support.

champikasam, 2018-08-16:
I was looking for this feature. Any idea when this will be available in builds? Thanks!

vaidyasr, 2018-08-24:
Will that be upgraded via web page of existing firmware?.
And will there be an option to change the no. of switches in admin page?

xoseperez, 2018-08-25:
Yes, it can be upgraded from the web UI and yes, there will be (not yet but soon) the option to change the number of switches from the web UI.

closed 2018-08-25 by xoseperez

peterhoeg, 2018-11-14:
While changing ```DUMMY_RELAY_COUNT``` in ```custom.h``` makes additional 8 relays available in the UI, only the first 8 are submitted to home assistant for auto-discovery.

I also noticed, that the newly added additional 8 relays would have a garbage ```1DE200F002E41C45E8``` entry pre-populated in the "Switch OFF" fields for relays 8 to 15.

peterhoeg, 2018-11-14:
Should I open new issues for these 2 and reference back to this?

mcspr, 2018-11-14:
@peterhoeg For rfbridge see https://github.com/xoseperez/espurna/issues/1308.
Hass issue is different, but close. And it might also just lose the data in transport, requiring some delay.

vogelfreiheit, 2018-11-14:
Yeah, #1308 is the same issue behind peterhoeg's problem, but it seems like in his case the data comes from uninitialized memory (hence why it is "garbage" in his case). Possibly an index programming error somewhere, in some loop where the logic wraps around the hardcoded old limit or some such. Not sure why the issue is closed. @xoseperez  could you take a look? Since this will likely manifest in potentially destructive ways (whenever a write operation occurs it's going to be out of bounds too, thrashing whatever is there most likely).