xoseperez/espurna #1139: [Question] Posibbility to add physical button to Magic Home LED cotroller.

kpcz, 2018-08-16:
Hi. Flashed latest Espruna (espurna-1.13.1-magichome-led-controller.bin) to v.1.1 Magic Home controller, everything works great via web gui, and via Home Assistant via MQTT. 

Now I am struggling to add an physical button to turn on or off the Light and also so that it would be possible to turn in back on via home assistant.  Is it some how possible via GPIO, or should I forget the idea? :(

looking forward for help. Thanks.

controller looks like this https://ibb.co/f5EUi9

kpcz, 2018-08-16:
I am eager to try, one more question, how to define what device I will be compile to in hardware.h file, as there is list of all devices?
Thank You!

Valcob, 2018-08-16:
Check the MAGICHOME_LED_CONTROLLER(search this string in file)  defenition. Note that this one uses most of the pins ;) so you are left with not so many of them. IO4 is IR IO5 is  Green IO14 is red IO12 is blue and IO13 is white

kpcz, 2018-08-16:
yes, added the configuration to the hardware.h, to the MAGICHOME_LED_CONTROLLER, but when I compile, I get error:

```
sketch\config/hardware.h:2608:6: error: #error "UNSUPPORTED HARDWARE!!"
     #error "UNSUPPORTED HARDWARE!!"
```
i think i need somewhere to define that I am compiling for MAGICHOME_LED_CONTROLLER or am I missing something..

Valcob, 2018-08-16:
Yep, that is done in arduino.h just find the line and uncomment it

kpcz, 2018-08-16:
yup, just found it as you typed :) Thanks. Will try now

kpcz, 2018-08-16:
cannot compile it, all the time "No such file or directory" on includes. tried to gather them all, still no luck.

Valcob, 2018-08-17:
Looks like you didn't read the wiki ;)
You can start here https://github.com/xoseperez/espurna/wiki/ArduinoIDE

brgaulin, 2018-08-19:
(sorry to ask here, no chat in github) @kpcz I was going to flash a V1.1(same board as you pictured) tomorrow. But when I looked at the [build files](https://github.com/xoseperez/espurna/blob/4f61d3aa7e2e6526fdb518a9f84785b055ac0ab7/code/espurna/config/hardware.h#L1045-L1101), I saw a magic home, and a magic home v2 build. The Magic Home build v1 references the Red channel being on GPIO14, but the ESP8285 GPIO14 isn't soldered on the V1.1(see image below, GPIO14 would be the 5th notch on the top)
[image]

So my curious question is: Did the V1 magic home binary work for you ok? And are your reds/greens ok? I'll also try and report back tomorrow after I spend some time on this. But the v2 build seems to make more sense. I just grabbed a multimeter and the IO pins seem to match up(ex. GPIO5 was red, 12 was green, 13 was blue, 15 was white)

kpcz, 2018-08-19:
hi @brgaulin I was only interested in using only single channel LED, so i did not bother to get working the colored LED strip. But I did test the RGBWW led strip with both v1 and v20 binary and none of them worked out of box correctly for all lights. So I guess you will have to experiment with hardware.h file to get the pins correct. :)

And yes, you can also get the physical push-button functionality, as it is described above, it works :)

Valcob, 2018-08-19:
@kpcz glad to hear that the button worked. Regarding pull up resistor that means that the pin logically(and physically through a resistor) is pulled to a HIGH state ( more on this you can check here [Pull up resistor tutorial](https://www.youtube.com/watch?v=wxjerCHCEMg) )
Now if everything is working we can consider this issue resolved so could you please close it?
Thanks.

closed 2018-08-19 by kpcz