xoseperez/espurna #1140: RF bridge with ESPURNA version 1.13.1 stopped sending and receiving RF signals.

DmitrySvyaznoy, 2018-08-18:
RF bridge with ESPURNA version 1.13.1 stopped sending and receiving RF signals.
The problem arose after 1 month of using the version of ESPURNA 1.13.1.
After reboot, RF Bridge resumes receiving and sending RF commands, but after 2-3 minutes it stops doing it.

There are no errors in the debug console.

[619406] [MQTT] Sending home/status => 1 (PID 1)
[619513] [MQTT] Sending home/data => {"app":"ESPURNA","version":"1.13.1","board":"ITEAD_SONOFF_RFBRIDGE","host":"ESPURNA-CAD038","ip":"192.168.8.199","mac":"DC:4F:22:CA:D0:38","rssi":"-43","uptime":"619","datetime":"2018-08-18 16:33:05","freeheap":"15496","relay/0":"0","relay/1":"0","relay/2":"0","relay/3":"0","relay/4":"0","relay/5":"0","relay/6":"0","relay/7":"0","vcc":"3191","loadavg":"1","time":"2018-08-18 16:33:06","id":1094432006} (PID 1)
[919409] [MQTT] Sending home/status => 1 (PID 1)
[919516] [MQTT] Sending home/data => {"app":"ESPURNA","version":"1.13.1","board":"ITEAD_SONOFF_RFBRIDGE","host":"ESPURNA-CAD038","ip":"192.168.8.199","mac":"DC:4F:22:CA:D0:38","rssi":"-44","uptime":"919","datetime":"2018-08-18 16:38:05","freeheap":"15496","relay/0":"0","relay/1":"0","relay/2":"0","relay/3":"0","relay/4":"0","relay/5":"0","relay/6":"0","relay/7":"0","vcc":"3191","loadavg":"1","time":"2018-08-18 16:38:06","id":1094432007} (PID 1)
[219412] [MQTT] Sending home/status => 1 (PID 1)
[219519] [MQTT] Sending home/data => {"app":"ESPURNA","version":"1.13.1","board":"ITEAD_SONOFF_RFBRIDGE","host":"ESPURNA-CAD038","ip":"192.168.8.199","mac":"DC:4F:22:CA:D0:38","rssi":"-45","uptime":"1219","datetime":"2018-08-18 16:43:05","freeheap":"15496","relay/0":"0","relay/1":"0","relay/2":"0","relay/3":"0","relay/4":"0","relay/5":"0","relay/6":"0","relay/7":"0","vcc":"3190","loadavg":"1","time":"2018-08-18 16:43:06","id":1094432008} (PID 1)
[519415] [MQTT] Sending home/status => 1 (PID 1)
[519522] [MQTT] Sending home/data => {"app":"ESPURNA","version":"1.13.1","board":"ITEAD_SONOFF_RFBRIDGE","host":"ESPURNA-CAD038","ip":"192.168.8.199","mac":"DC:4F:22:CA:D0:38","rssi":"-44","uptime":"1519","datetime":"2018-08-18 16:48:05","freeheap":"15496","relay/0":"0","relay/1":"0","relay/2":"0","relay/3":"0","relay/4":"0","relay/5":"0","relay/6":"0","relay/7":"0","vcc":"3190","loadavg":"1","time":"2018-08-18 16:48:06","id":1094432009} (PID 1)
[819418] [MQTT] Sending home/status => 1 (PID 1)
[819525] [MQTT] Sending home/data => {"app":"ESPURNA","version":"1.13.1","board":"ITEAD_SONOFF_RFBRIDGE","host":"ESPURNA-CAD038","ip":"192.168.8.199","mac":"DC:4F:22:CA:D0:38","rssi":"-44","uptime":"1819","datetime":"2018-08-18 16:53:05","freeheap":"15496","relay/0":"0","relay/1":"0","relay/2":"0","relay/3":"0","relay/4":"0","relay/5":"0","relay/6":"0","relay/7":"0","vcc":"3190","loadavg":"1","time":"2018-08-18 16:53:06","id":1094432010} (PID 1)
[820709] [NTP] UTC Time  : 2018-08-18 13:53:07
[820711] [NTP] Local Time: 2018-08-18 16:53:07
[119422] [MQTT] Sending home/status => 1 (PID 1)
[119528] [MQTT] Sending home/data => {"app":"ESPURNA","version":"1.13.1","board":"ITEAD_SONOFF_RFBRIDGE","host":"ESPURNA-CAD038","ip":"192.168.8.199","mac":"DC:4F:22:CA:D0:38","rssi":"-46","uptime":"2119","datetime":"2018-08-18 16:58:05","freeheap":"15496","relay/0":"0","relay/1":"0","relay/2":"0","relay/3":"0","relay/4":"0","relay/5":"0","relay/6":"0","relay/7":"0","vcc":"3190","loadavg":"1","time":"2018-08-18 16:58:05","id":1094432011} (PID 1)
[389206] [WEBSOCKET] Requested action: relay
[389208] [RELAY] #2 scheduled ON in 0 ms
[389211] [RELAY] #2 set to ON
[389215] [RFBRIDGE] Enqueuing MESSAGE '3B1A01E005B4F055F0' 1 time(s)
[389319] [MQTT] Sending home/data => {"relay/2":"1","time":"2018-08-18 17:02:35","mac":"DC:4F:22:CA:D0:38","host":"ESPURNA-CAD038","ip":"192.168.8.199","id":1094432012} (PID 1)
[390217] [RELAY] Saving mask: 4
[390764] [WEBSOCKET] Requested action: relay
[390767] [RELAY] #2 scheduled OFF in 0 ms
[390771] [RELAY] #2 set to OFF
[390774] [RFBRIDGE] Enqueuing MESSAGE '44C00244069A260219' 1 time(s)
[390878] [MQTT] Sending home/data => {"relay/2":"0","time":"2018-08-18 17:02:37","mac":"DC:4F:22:CA:D0:38","host":"ESPURNA-CAD038","ip":"192.168.8.199","id":1094432013} (PID 1)
[391776] [RELAY] Saving mask: 0
[397539] [WEBSOCKET] Requested action: relay
[397542] [RELAY] #5 scheduled ON in 0 ms
[397546] [RELAY] #5 set to ON
[397549] [RFBRIDGE] Enqueuing MESSAGE '32B401A404F631C0C0' 1 time(s)
[397654] [MQTT] Sending home/data => {"relay/5":"1","time":"2018-08-18 17:02:43","mac":"DC:4F:22:CA:D0:38","host":"ESPURNA-CAD038","ip":"192.168.8.199","id":1094432014} (PID 1)
[398552] [RELAY] Saving mask: 32
[399475] [WEBSOCKET] Requested action: relay
[399478] [RELAY] #5 scheduled OFF in 0 ms
[399483] [RELAY] #5 set to OFF
[399588] [MQTT] Sending home/data => {"relay/5":"0","time":"2018-08-18 17:02:45","mac":"DC:4F:22:CA:D0:38","host":"ESPURNA-CAD038","ip":"192.168.8.199","id":1094432015} (PID 1)
[400486] [RELAY] Saving mask: 0
[419424] [MQTT] Sending home/status => 1 (PID 1)
[419531] [MQTT] Sending home/data => {"app":"ESPURNA","version":"1.13.1","board":"ITEAD_SONOFF_RFBRIDGE","host":"ESPURNA-CAD038","ip":"192.168.8.199","mac":"DC:4F:22:CA:D0:38","rssi":"-45","uptime":"2419","datetime":"2018-08-18 17:03:05","freeheap":"15528","relay/0":"0","relay/1":"0","relay/2":"0","relay/3":"0","relay/4":"0","relay/5":"0","relay/6":"0","relay/7":"0","vcc":"3190","loadavg":"3","time":"2018-08-18 17:03:05","id":1094432016} (PID 1)
[719428] [MQTT] Sending home/status => 1 (PID 1)
[719534] [MQTT] Sending home/data => {"app":"ESPURNA","version":"1.13.1","board":"ITEAD_SONOFF_RFBRIDGE","host":"ESPURNA-CAD038","ip":"192.168.8.199","mac":"DC:4F:22:CA:D0:38","rssi":"-46","uptime":"2719","datetime":"2018-08-18 17:08:05","freeheap":"15496","relay/0":"0","relay/1":"0","relay/2":"0","relay/3":"0","relay/4":"0","relay/5":"0","relay/6":"0","relay/7":"0","vcc":"3190","loadavg":"1","time":"2018-08-18 17:08:05","id":1094432017} (PID 1)

DmitrySvyaznoy, 2018-08-18:
info:

[683035] [INIT] ESPURNA 1.13.1
[683036] [INIT] xose.perez@gmail.com
[683037] [INIT] http://tinkerman.cat

[683038] [INIT] CPU chip ID: 0xCAD038
[683039] [INIT] CPU frequency: 80 MHz
[683041] [INIT] SDK version: 1.5.3(aec24ac9)
[683045] [INIT] Core version: 2.3.0
[683048] [INIT] Core revision: 
[683050] 
[683052] [INIT] Flash chip ID: 0x144051
[683058] [INIT] Flash speed: 40000000 Hz
[683060] [INIT] Flash mode: DOUT
[683061] 
[683063] [INIT] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[683071] [INIT] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[683077] [INIT] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[683084] [INIT] Firmware size       :   469136 bytes /  115 sectors (   1 to  115)
[683091] [INIT] Max OTA size        :   552960 bytes /  135 sectors ( 116 to  250)
[683098] [INIT] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[683106] [INIT] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[683113] 
[683114] [INIT] EEPROM sectors: 251, 250
[683117]

xoseperez, 2018-08-26:
I see there are some messages missing there. After each switch toggling it should output a ```[RFBRIDGE] Action 0xA0``` message which is the ACK from the secondary microcontroller. This means the EFM is not ACK, so probably not receiving the messages or unable to parse them...

Another thing I have noticed is that the message is only enqueued once. Did you change the `RF_SEND_TIMES` value to 1?

padcom, 2018-08-27:
I am experiencing the exact same problem with firmware built from the latest sources. I needed to revert to 1.13.1 to get it back up and running.

It'd be great if that would get fixed - there are some changes I'd like to make to the firmware in the direct version to be able to read temperature sensors and other RF equipment.

DmitrySvyaznoy, 2018-08-28:
In general settings, I set "Double click delay" 1000 ms, since the RF door sensor sends 3 pulses in a row when triggered.

Now installed on one of its RF Bridge version 1.13.2.

I observe behavior.