xoseperez/espurna #1142: Sensors Tab in Web UI Missing

derlucas, 2018-08-19:
The "Sensors" Tab in Web UI is missing since commit 31436ead43b778705802a81a9b3781094214d19e

xinuxnt, 2018-08-19:
Hello, Did you uncommented the sensor on "arduino.h"?

derlucas, 2018-08-20:
No i did not change any code. I just used this command to build the firmware for my purpose:

`ESPURNA_FLAGS="-DMHZ19_SUPPORT=1" pio run -e wemos-d1mini -t upload --upload-port /dev/ttyUSB`

xoseperez, 2018-08-20:
The problem is that the WEBUI_IMAGE definition section in deoendencies.h
must be after the sensor.h loading...

I cannot change it now (won't have a laptop with me until next week) but
maybe someone can test it and do a OR

xoseperez, 2018-08-20:
That alone won't work since the sensors are enabled in the sensors.h file.

Valcob, 2018-08-20:
nope I guess it will )  ESPURNA_FLAGS="-DMHZ19_SUPPORT=1"

xoseperez, 2018-08-26:
Fixed in `dev` branch. Will be released soon.

closed 2018-08-26 by xoseperez

schweini, 2018-12-04:
I just compiled and uploaded dev, and unless I'm doing something wrong, still can't see the Sensor calibration panel.

mcspr, 2018-12-05:
@schweini does running `info` in the debug panel show `[MAIN] WebUI image: SENSOR`?
as mentioned above - correct file will be included when `SENSOR_SUPPORT == 1`