xoseperez/espurna #1143: Counter sensor won't compile with EVENTS_INTERRUPT_MODE=BOTH

ways, 2018-08-20:
Hi

I want to use a reed sensor, and am experimenting with a counter sensor to get instant notifications. The docs says:

> EVENTS_INTERRUPT_MODE | On which signal transition to react. 
> One of:  - RISING (default)  - FALLING  - BOTH

How ever when I compile with:

> build_flags = ${common.build_flags_1m0m} -DITEAD_S20 -DEVENTS_SUPPORT=1 -DEVENTS_PIN=1 -DEVENTS_PIN_MODE=INPUT_PULLUP -DEVENTS_INTERRUPT_MODE=BOTH

I get:

```Building firmware images...
...
/home/user/Downloads/git/espurna/code/espurna/sensor.ino: In function 'void _sensorLoad()':
<command-line>:0:23: error: 'BOTH' was not declared in this scope
/home/user/Downloads/git/espurna/code/espurna/sensor.ino:454:34: note: in expansion of macro 'EVENTS_INTERRUPT_MODE'
sensor->setInterruptMode(EVENTS_INTERRUPT_MODE);
^
*** [.pioenvs/itead-s20/src/espurna.ino.cpp.o] Error 1
================================= [ERROR] Took 3.76 seconds =================================
```

I tried looking at the code, but it's beyond me to figure out how or where BOTH should be defined.

Very grateful for the project! =)

mcspr, 2018-08-20:
Hello.

I think comment meant CHANGE not BOTH. See https://arduino-esp8266.readthedocs.io/en/latest/reference.html#digital-io and https://github.com/esp8266/Arduino/blob/18f643c7e2d6a0da9d26ff2b14c94e6536ab78c1/cores/esp8266/Arduino.h#L75-L78

ways, 2018-08-21:
Ah, thanks.

PR #1144 (2018-08-25): Fix _INTERRUPT_MODE comment
Resolves #1143
closed 2018-08-21 by ways