xoseperez/espurna #1145: Switch log in domoticz reports state change twice

miroslavpetrov, 2018-08-21:
Hello,

I am using domoticz 4.97 and espurna 1.13.1 on sonoff duals connected to a mqtt broker. When I change the state of a switch in domoticz the log of the switch reports the state change twice. I guess this is because domoticz sends a message in domoticz/out topic and then the sonoff reports the state change in domoticz/in topic. See the screenshot below. Do you have any idea on how to fix this?
[image]

Camposmansi, 2018-09-26:
Good, the same thing happens to me, I send from domoticz to espurna and espurna returns this value to home automation, which makes me a double change of state and a double message
Since I use it happens to me and I can not find how to solve it, what I found in the debug ...

[893767] [DOMOTICZ] Received value 1 for IDX 108
[893769] [RELAY] #0 scheduled ON in 0 ms
[893771] [RELAY] #0 set to ON
[893784] [MQTT] Sending domoticz/in => {"idx": 108, "nvalue": 1, "svalue": ""} (PID 1)

Camposmansi, 2018-09-26:
by the way, with home assistant does not happen

loglux, 2018-09-30:
It happens in Domoticz with ESPurna running firmware on Sonoff POW, Sonoff S20, Sonoff RF Bridge. I have the same POW switch with Tasmota, that does not happen.

Camposmansi, 2018-09-30:
Yes, I just tried it with tasmota and it does not happen.
I've tried all kinds of settings and I do not know why, do I have to wait for an update?

loglux, 2018-09-30:
I think so. It is obvious. We should wait for 1.14

loglux, 2018-09-30:
Along with this I can see On On and then Off Off in a Switch's statistics in Domoticz at the same time, for example at 07:25:41 but Domoticz log show that nothing had happened at this time.

loglux, 2018-09-30:
it's maybe a time synchronisation issue as well

Camposmansi, 2018-09-30:
If they do not fix it I'm sorry but I have to leave ESPurna and pass to tasmota, but I love ESPurna and I would be very sorry

Camposmansi, 2018-09-30:
> Along with this I can see On On and then Off Off in a Switch's statistics in Domoticz at the same time, for example at 07:25:41 but Domoticz log show that nothing had happened at this time.

Yes of course, the same thing happens to me but only in domoticz in home assistant does not happen

loglux, 2018-09-30:
Trust me, Tasmota is a bigger crap. It can show that POW is Off while it's continuing to be On.

loglux, 2018-09-30:
be patient

ColinShorts, 2018-09-30:
@loglux  it might be worth mentioning the corresponding issue at domoticz... https://github.com/domoticz/domoticz/issues/2175

Camposmansi, 2018-10-01:
The problem they have is the same but it happens to me through MQTT and not through API

Camposmansi, 2018-10-03:
Is this close and not found

JPasa, 2019-02-09:
Hello, this error persists.

I am using domoticz 4.97 and espurna 1.13.1 on sonoff duals connected to a mqtt broker. When I change the state of a switch in domoticz the log of the switch reports the state change twice.

xoseperez, 2019-02-10:
This has been fixed with #1536. Will be released with 1.13.4.

closed 2019-02-10 by xoseperez