xoseperez/espurna #1151: SoftAP name not set properly

hyteoo, 2018-08-23:
When in soft AP, it seems that the hostname and  is ignored.
Ex. 
[image]
[image]

@mcspr 
> ESP_... comes from justwifi lib: https://github.com/xoseperez/justwifi/blob/c423f102d0ece574beac6b536f11cb692709763c/src/JustWifi.cpp#L52
If no SSID provided, it uses hostname: https://github.com/xoseperez/justwifi/blob/c423f102d0ece574beac6b536f11cb692709763c/src/JustWifi.cpp#L296-298

> As Xose did mention, something funky must be happening to the settings. Only thing that makes sense if hostname is empty here / `hostname `key from eeprom settings returns nothing.

https://github.com/xoseperez/espurna/blob/4f61d3aa7e2e6526fdb518a9f84785b055ac0ab7/code/espurna/wifi.ino#L36-L40

> And looking at wifi listings only common thing I did notice that in all cases ESP APs are passwordless, which is likely happening because setSoftAP returns early with some condition failing.

mcspr, 2018-08-24:
@hyteoo tnx :) #1103 did mention long hostname... and i did mention justwifi as a source but that's not it. this is just sdk doing which happens to have the same ssid creating logic.

Still, from justwifi: `WiFi.softAP` is called at some point and it will enable AP first and only then check / set given ssid+passphrase. If hostname is longer than 31 chars or pass is less than 8 i too can see this behaviour. e.g. erasing flash (sdk does persist softap settings) and adding '-DHOSTNAME="..."' or '-DADMIN_PASS="test"' results in the same ESP_XXXXXX ssid.

WebUI does enforce hostname / password lengths.
`code/espurna/config/general.h` should mention this limitation for both HOSTNAME and ADMIN_PASS.
And there must be 3rd layer of length checks so debug log shows this too, otherwise this is confusing :/

hyteoo, 2018-08-24:
I checked this and indeed, the password was the problem. In my case it had 7 characters.
I had it setup in my general.h define, so I never changed it from the webui.
After checking the webUI, it seems that that is also wrong, as it suggests a minimum of 5 characters.
Therefore I propose the following:

1. Add a comment for the password define, warning it should be at least 8 characters (maybe even a preprocessor validation...even though I'm not sure you can do a strlen there....)
https://github.com/xoseperez/espurna/blob/4f61d3aa7e2e6526fdb518a9f84785b055ac0ab7/code/espurna/config/general.h#L12-L14

2. Change the WebUI text and validation from 5 to 8.

closed 2018-10-31 by stale[bot]