xoseperez/espurna #1152: RF Bridge add a push button

champikasam, 2018-08-24:
Hi,

I am using RF Bridge to connect a 433 MHz door sensor to Home Assistant. The sensor can send 4 different codes: door open, door close, tamper and low battery.

I have currently assigned door open and door close codes into a single dummy switch, and its working well with Home Assistant. But i'll need a push button rather than a toggle button to integrate tamper and low battery signals.

Default configuration in hardware.h only has a setting to set the number of dummy switches. Can i add a few push buttons into it as well ?

Thanks

hyteoo, 2018-08-24:
@champikasam off topic... can you post a link from where you got the door sensor?

regarding your question, in the Switches tab you can do that. Just set Pulse mode and Pulse time. When the associated code comes in, it will set the button to the value specified in Pulse mode for <Pulse time> seconds (you can also use 0.1 there for ex.).
That will make the button act like a push button.
Hope that helps.

champikasam, 2018-08-30:
Thanks! 

I'm using this sensor: https://www.aliexpress.com/item/6pcs-lot-New-433mhz-While-Wireless-Home-Alarm-Window-Door-Sensor-to-Detect-Open-Door/32463726744.html?transAbTest=ae803_1&priceBeautifyAB=0

jsponz, 2018-09-11:
@champikasam how is the status with this? Could you confirm it works? How do you manage the four codes? 

Thanks!