xoseperez/espurna #1154: Web UI console usage

cnieves1, 2018-08-24:
Hi,
How can I use the web UI console usage on a 1.13.1 FW?
I found the command list at:
  https://github.com/xoseperez/espurna/wiki/Terminal
but I get "Unknown" for everything I type:
  17:34:05 CMD: HELP
  17:34:05 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
  17:34:07 CMD: help\n
  17:34:07 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
  17:34:11 CMD: HELP\N
  17:34:11 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
  17:34:13 CMD: help

Any hints? TIA!

xoseperez, 2018-08-25:
Not sure about this. That log is not from ESPurna, I think it's Tasmota...

cnieves1, 2018-08-25:
You were right! I used SonOTA to flash the S20, and it seems it didn't get the Espurna FW I downloaded...

closed 2018-08-25 by cnieves1