xoseperez/espurna #1159: Problem learning code using mosquitto_pub and rfin?

VaraI, 2018-08-26:
Hi,

 Thanks for the awesome espurna software. I'm still quite a newbie here. I flashed my sonoff RF bridge last February 2018 (using ESPEasy_v2.0.0-dev12). At first, I was able to learn RF codes from various 433mHz 1527 RF devices (motion sensor, door, fire alarm) using the following line in my raspberry pi running hassbian:

**mosquitto_pub -d -u MYUSERNAME -P MYPASSWORD -t snbridge/rfin**

(I was able to get the result like **"1E78xxxxxxxxx1335"** after hitting Enter the above line and triggering the rf sensor)

However, several months ago, I did some upgrade (sudo apt-get update & sudo apt-get upgrade) and this line (mosquitto_pub -d -u MYUSERNAME -P MYPASSWORD -t snbridge/rfin) is giving me the following error:

**Error: Both topic and message must be supplied.**

Does anyone have the same problem? Do I need to change the way I put in the command line? Please suggest how can I display the RF code in the command line.

Note: I found that learning code from espurna web interface isn't always a success. I'll get 2 beeps sound but no code showed up in the box. Using command line, I always get the RF code right away. But now the command line above doesn't work anymore.

**thank you in advance!**

xoseperez, 2018-08-26:
You are using `mosquitto_pub` which is the tool to publish messages, so you have to add both the topic and the payload (using the `-m` option). I guess you actually want to use `mosquitto_sub` to subscribe to a certain topic...

VaraI, 2018-09-01:
Thank you so much! I've just got a chance to try and it works perfectly.

closed 2018-09-02 by xoseperez