xoseperez/espurna #1163: No Relays and Alexa Support crashes WiFi or Device

CrazyIvan359, 2018-08-27:
Yesterday I spent most of the afternoon trying to get a Wemos D1 Mini working that will only have sensors connected to it. The device would boot, but not discover any networks, or broadcast one successfully.

So initially I built the device with all of my sensors (6 DS18B20s on separate wires, an HC-SR04, and I2C enabled) and serial debugging off and it didn't come up on my WiFi, nor could I see a hot spot being broadcast. Built it again and turned serial debugging on, it boots fine, doesn't find any networks upon scanning, and then says its in AP mode but I still can't see its network. Many builds later, removing more and more things, I end up with a build with absolutely nothing in it, the only things left in my device profile are manufacturer and board, and now it crashes! Turns out that was a good thing. It would boot and then start scanning for networks and crash, giving me a stack trace. Unfortunately I don't have said stack trace anymore, but I can repeat what I did and get it for you. Stack trace led me to Alexa, I didn't look into the code, but something in the Alexa module seems to crash or hang the WiFi stack if there are no relays or lights defined. Built again with nothing and Alexa disabled and its working perfectly, added back all sensors and all is still well.

TLDR: Alexa support does not work when the device has no relays or lights defined

xoseperez, 2018-08-27:
Can you send me the configuration in hardware.h of the device you are building? I have tried disabling all relays in my test board and Alexa did not crashed it...

CrazyIvan359, 2018-08-27:
Here is what I managed to get it stable with first:
(I had commented the remainder out, but removed here for size)
```
#elif defined(WEMOS_D1_MINI_MURPHYSENSOR)

    // Info
    #define MANUFACTURER        "WEMOS"
    #define DEVICE              "D1_MINI_MURPHYSENSOR"

    // Remove unneed things for space
    #define SCHEDULER_SUPPORT   0
    #define DOMOTICZ_SUPPORT    0
```

This is the final version I will be using. I have modified the GPIO limit (normally `<=15`) and added code for the additional Dallas sensors, which isn't causing any problems, though I don't have them working yet but that's a hardware issue.
```
#elif defined(WEMOS_D1_MINI_MURPHYSENSOR)

    // Info
    #define MANUFACTURER        "WEMOS"
    #define DEVICE              "D1_MINI_MURPHYSENSOR"

    // Remove unneed things for space
    #define SCHEDULER_SUPPORT   0
    #define ALEXA_SUPPORT       0             // MUST REMOVE WHEN DEVICE HAS NO RELAYS OR LIGHTS
    #define DOMOTICZ_SUPPORT    0

    // LEDs
    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    1

    // Disable serial, we're using the pins
    //#define DEBUG_SERIAL_SUPPORT    0

    // Disable sleep, there's a lot to do
    #define LOOP_DELAY_TIME     1              // Delay for this millis in the main loop [0-250]

    // Sensors
    #define SENSOR_PUBLISH_ADDRESSES 1
    #define DALLAS_SUPPORT      1
    #define DALLAS_PIN          14
    #define DALLAS_PIN2         12
    #define DALLAS_PIN3         13
    #define DALLAS_PIN4         5
    #define DALLAS_PIN5         4
    #define DALLAS_PIN6         0

    #define I2C_SUPPORT         1
    #define I2C_SDA_PIN         1
    #define I2C_SCL_PIN         3
    #define BMX280_SUPPORT      1
    #define BMX280_ADDRESS      0x76    // 0x00 means auto
    #define BMX280_ADDRESS2     0x77

    #define HCSR04_SUPPORT      1
    #define HCSR04_TRIGGER      15
    #define HCSR04_ECHO         16      // library does not use pullup so this works
```

Side note: Will the BMX280 bus search find both sensors, or do I need to specify both addresses as I have done?

CrazyIvan359, 2018-08-27:
Oh, and I just remembered I'm using 1.13.0 still, not sure if you made any changes that would affect this since then.

CrazyIvan359, 2018-08-27:
I can flash a spare device I have when I get home and upload serial logs or do other testing.

The device would still work when I was building with sensors, it was only when it was completely empty that I wouldn't even get anything on the serial. I get the impression that its not actually crashing, but hanging, all of the resets were caused by the watchdog

xoseperez, 2018-08-27:
OK, the problem is in the fauxmoESP library, it does not handle correctly the case when there are no devices defined. I have changed it in the 2.4.X branch of the library. 

But ESPurna `dev` uses the new 3.0.0 version which emulates Philips Hue bulbs instead of WeMo plugs. This new version does not suffer from the same bug.

CrazyIvan359, 2018-08-27:
Well I should have stuck my nose in alexa.ino sooner, I found our problem
```
    // Lights
    #if RELAY_PROVIDER == RELAY_PROVIDER_LIGHT

        // Global switch
        alexa.addDevice(hostname.c_str());

        // For each channel
        for (unsigned char i = 1; i <= lightChannels(); i++) {
            alexa.addDevice((hostname + " " + i).c_str());
        }

    // Relays
    #else

        unsigned int relays = relayCount();
        if (relays == 1) {
            alexa.addDevice(hostname.c_str());
        } else {
            for (unsigned int i=1; i<=relays; i++) {
                alexa.addDevice((hostname + " " + i).c_str());
            }
        }

    #endif
```
This looks like it will error if `relayCount()` returns 0

CrazyIvan359, 2018-08-27:
Given that the Alexa library is used for control, perhaps excluding the library at build if no relays or lights are defined, would be a cleaner solution? It would save on RAM and processor cycles as well

CrazyIvan359, 2018-08-27:
Say in dependencies.h we put something like this:
```
#if DUMMY_RELAY_COUNT == 0 || (RELAY1_PIN == GPIO_NONE && RELAY2_PIN == GPIO_NONE && RELAY3_PIN == GPIO_NONE && RELAY4_PIN == GPIO_NONE && RELAY5_PIN == GPIO_NONE && RELAY6_PIN == GPIO_NONE && RELAY7_PIN == GPIO_NONE && RELAY8_PIN == GPIO_NONE)
  #define ALEXA_SUPPORT 0
#endif
```

xoseperez, 2018-08-27:
Future versions of ESPurna will allow adding relays on the fly, so Alexa would still be useful...

CrazyIvan359, 2018-08-27:
Oh yes, I remembering seeing that on here somewhere that you were working on that. Definitely looking forward to on-the-fly hardware configuration!

Valcob, 2018-08-28:
@xoseperez  There is a potential segfault exception in the fauxmo  library in 
```
char * fauxmoESP::getDeviceName(unsigned char id, char * device_name, size_t len) {
    if (0 <= id && id <= _devices.size() /* && device_name */) { // we should add  a null check for device name
        /*-->*/ strncpy(device_name, _devices[id].name, len); // if device_name is null we have a seg fault
    }
    return device_name;
}
```

xoseperez, 2018-08-29:
@Valcob Thank you. Updated fauxmoESP lib to 3.0.1 and ESPurna reference.

closed 2018-09-02 by xoseperez