xoseperez/espurna #1168: WPS not works again

serra82, 2018-08-29:
Thank you @xoseperez for wps fix, but the problem persists.
this is the log of AccessPoint:
19:32:32 wireless,info wlan1: WPS virtual button pushed 
19:32:35 wireless,info 5C:CF:7F:A0:3C:0C@wlan1: connected 
19:32:40 wireless,info 5C:CF:7F:A0:3C:0C@wlan1: disconnected, unicast key exchange timeout

a correct WPS association should be like this:
19:40:07 wireless,info wlan1: WPS virtual button pushed  
19:40:09 wireless,info wlan1: WPS association from B4:B6:76:03:B1:64 
19:40:09 wireless,info B4:B6:76:03:B1:64@wlan1: connected 
19:40:09 wireless,info wlan1: WPS of B4:B6:76:03:B1:64 started, associated 
19:40:09 wireless,info wlan1: WPS of B4:B6:76:03:B1:64, received info: 'DESKTOP-0R5SPTF'  
19:40:09 wireless,info wlan1: WPS of B4:B6:76:03:B1:64, do registration 
19:40:09 wireless,info wlan1: WPS of B4:B6:76:03:B1:64 complete 
19:40:09 wireless,info wlan1: WPS button reset 
19:40:09 wireless,info B4:B6:76:03:B1:64@wlan1: disconnected, WPS complete 
19:40:10 wireless,info B4:B6:76:03:B1:64@wlan1: connected 
19:40:11 dhcp,info dhcp1 assigned 192.168.88.149 to B4:B6:76:03:B1:64 

in addition now if I try two times the wps association, the espurna crashs:

WIFI.WPS
[218132] [WIFI] Access point destroyed
[218132] [WIFI] Disconnected
[218133] +OK
_state: 7, WiFi.getMode(): 0
[218133] [WIFI] WPS started
_state: 8, WiFi.getMode(): 1
_state: 9, WiFi.getMode(): 1
[233544] [WIFI] Creating access point
[233545] [WIFI] Captive portal enabled
[233546] [MDNS] OK
[233546] [WIFI] WPS failed
_state: 15, WiFi.getMode(): 1
_state: 0, WiFi.getMode(): 1
WIFI.WPS
[242674] [WIFI] Access point destroyed
[242882] [WIFI] Disconnected
[242882] +OK
_state: 7, WiFi.getMode(): 0
[242882] [WIFI] WPS started
_state: 8, WiFi.getMode(): 1

Exception (3):
epc1=0x4000deea epc2=0x00000000 epc3=0x00000000 excvaddr=0x4023433e depc=0x00000000

ctx: sys 
sp: 3ffffd00 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffea0:  3ffebbfc 000009d4 000009d4 4010020c  
3ffffeb0:  4024843f 00000030 00000014 ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  00000000 00000000 00000000 00000000  
3ffffee0:  00000000 00000000 00000000 00000000  
3ffffef0:  00000000 00000000 00000000 00000000  
3fffff00:  00000000 00000000 00000000 00000000  
3fffff10:  00000000 00000000 00000000 00000000  
3fffff20:  00000000 00000000 0000002e 00000000  
3fffff30:  3fff5024 40247f6b 3ffeed60 3fff78a4  
3fffff40:  00000000 4023433e 3ffeed60 3ffebc4c  
3fffff50:  3ffebc4c 000000b8 00000000 0000002e  
3fffff60:  00000000 000000b8 4022e1bb 3ffeed60  
3fffff70:  3ffebc40 3fffdcc0 3ffeb4b0 3ffeb4b0  
3fffff80:  00000080 3ffeed60 00000000 3fff1550  
3fffff90:  4022dae7 3fffdab0 00000000 4022a613  
3fffffa0:  3ffeb4b0 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

xoseperez, 2018-08-29:
It's working fine with my router (not as sophisticated as yours but...).

mcspr, 2018-08-30:
@serra82 hello. can you try with latest SDK? see https://github.com/esp8266/Arduino/issues/3941
for platformio it is setting `platform = espressif8266@1.8.0` and adding `-D NO_EXTRA_4K_HEAP` to build_flags

serra82, 2018-08-30:
@mcspr with the setting that you suggest the espurna doesn't crash, but the wps problem permain.
@xoseperez I don't understand why with justWiFI the wps example works fine, anyway I'll test with a different router.
Thank you
Andrea

xoseperez, 2018-09-12:
It works for me. Both the library examples and in ESPurna. If using SDK 2.4.2 (PlatformIO espressif8266@1.8.0 platform) use the `-DNO_EXTRA_4K_HEAP` build flag as suggested by @mcspr .

xoseperez, 2018-09-13:
The latest version of the JustWifi library checks for WPS support under Core 2.4.2

closed 2018-09-25 by xoseperez