xoseperez/espurna #1173: Espurna Architecture - (discussion/suggestion)

zafrirron, 2018-09-01:
Espurna is a great project! (thanks to @xoseperez) the extensibility, configurability, and services support puts it at the top of 8266 frameworks for ESP projects.

Looking at its architecture (I love it...) I have few thoughts about possible future directions..
the main objective is making espura a true platform instead of inclusive project for all market possible 8266 based modules and optional sensors.... 

AS espurna currently support ESP8266 family (ESP32 hopefully), the basic esp8266 modules have very few configurations (ESP01(S),  ESP05, ESP12(E/F), the main difference with regards to espurna is memory size.
espurna "hardware" support is mainly different configuration of GPIO pins assignments.
looking at platformio.ini (build configuration)   most env definitions are identical (depends on board memory mostly) with additions of flags, manly to enable disable different services.

as a result building a binary load module for different 8266 module in the market could be avoided by having a core binary (maybe for different ESP flavours) and different config files.

since most GPIO configurations could be done in runtime (changing pins, button types, button behaviour, relay types...), they could be externalized (why shouldn't I be able to change button to switch without build? or assign lnglngclik hard reset).

Looking at sensors, the basic sensor functionality is different by HW logic types (serial, I2C...) so maybe put the H/W related logic and the different sensors should be a configuration only.
 
The main espurna structure could be:
1. Core Services (wifi, button, relay, ota, sensors H/W related logic) - core binay
2. External Services (mqtt, homeassistant, NTP...) - should be enabled disabled by configuration (I guess currently only ESP01 512k can not hold all in memory, so dynamic build should be considered.
3. GUI (supports core services, External Services....configuration selection...)
4. Plugin system (to enable extensibility for developers and creating a library of plugins)  
5. Boards configuration files (could be updated independently) 
6. Sensors configuration files  (could be updated independently) 

This proposal may change the espurna project model and will allow decoupling of platform related activity from unique configuration activities), to allow scaling of this great project (ESP32 ?).

splitting the platform parts will also enable better support (separate for each topic) and creating separate  development streams.  

as most espurna users are (should) not developers, the model should allow loading the same espurna binary without building and selecting/setting configuration (select commercial name of board flavour, sensor name...configure external services), developers should mainly build plugins (a very popular plugins should be considered as candidates for next release).

lobradov, 2018-09-01:
You might want to checkout 'v2' branch which started to go in the direction
you describe above, but it's still in very early stages.
Features are selected and compiled in during build time, but then you can
arrange them according to the configuration. Furthermore, actual hardware
is described in JSON files, so you could just upload the config and get
your exact board supported.

I just started looking at it and it seems that CLI based config is
available, but Web UI will need to be rewritten to support this "online
configuration".

zafrirron, 2018-09-01:
Thanks @lobradov, I will check it out.

xoseperez, 2018-09-02:
The idea behind `v2` is to do a step-by-step transition. That is why it still defined the hardware at build time, you can still build specific binaries for each device but then you can modify the hardware via CLI or uploading a JSON file and reloading or rebooting (not all settings are updated on reload).

The final stage would be to have only a few binaries (basic, dimmer, my92xx, hlw8012, sensor,...) and configuration files in JSON format for each device. 

Including a features in the binary has 2 consequences: bigger binary size (we can do two-step-updates but I still prefer keeping it under 500kb if possible) and init RAM which should be reduced to 0 if possible (some modules still init buffers at power on). With this two issue in mind, we can talk about adding features to the main binary.

tve, 2018-11-08:
Is V2 dead?