xoseperez/espurna #1175: neopixel support on the Generic RGBLed v1.0 using the fastled branch

ColinShorts, 2018-09-01:
Hi all, I've spent a couple of hours trying to get some neopixels working on this board to no avail. I appreciate that I'm working outside of the master/dev branch, but hoped someone would have had some experience with it.

I'm using (only) 2 pixels for testing, using WS2811 on pin 2, with a single dummy relay (copied/noticed from the example). I've managed to get the board and LEDs working with a couple of different arduino sketches, so reasonably confident that it should work. I'm building Espurna using platformio rather than Arduino IDE if that matters.

Occasionally the first LED lights up green, is it maybe a timing/frequency issue?

xoseperez, 2018-09-02:
The `fastled` branch is quite old and has not been merged with the latest code. I can't recall if it was in working state already... Might give it a try over the next days.

cnf, 2018-11-02:
i am trying to do the same thing, and I can't get anywhere on this. Is there any interest on getting this to work?

whoamicz, 2019-01-04:
Hello, I just started using this FW and I just want to give thumbs up for fastled strips support.

oscarsan1, 2019-04-02:
I wil try to integrate neopixel or fast led to Espurna in the next weeks for control led strip with hundreds of leds. Let's see if is it possible with Wifi interruptions while sending control bits. Any advance in this field?

xkidro, 2019-06-19:
Any progress on this ?