xoseperez/espurna #1178: OneWire DS18B20 temperature sensor

Danny89530, 2018-09-02:
Hi @xoseperez,
I'm trying to flash a NodeMCU Lolin with a temperature sensor and a Relay.
No problem with the relay but I cannot figure out how to get the sensor working.
These are my build flags:
```
[env:danny-thermostat]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board_4m}
board_build.flash_mode = ${common.flash_mode}
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags_4m1m}
              '-DDEVICE="THERMOSTAT"'
              '-DMANUFACTURER="DANNY"'
              -DDEBUG_FAUXMO=Serial
              -DNOWSAUTH
              -DLED1_PIN=2 -DLED1_PIN_INVERSE=1
              -DDALLAS_PIN=4 -DDALLAS_SUPPORT=1
              -DRELAY1_PIN=5 -DRELAY1_TYPE=0
              -DALEXA_SUPPORT=0 -DALEXA_ENABLED=0
              -DDOMOTICZ_SUPPORT=0 -DDOMOTICZ_ENABLED=0
              -DHOMEASSISTANT_SUPPORT=0 -DHOMEASSISTANT_ENABLED=0
              -DTHINGSPEAK_SUPPORT=0 -DTHINGSPEAK_ENABLED=0
              -DBUTTON_SUPPORT=0
upload_speed = ${common.upload_speed_fast}
monitor_speed = ${common.monitor_speed}
extra_scripts = ${common.extra_scripts}
```

Obviously the sensor was attached to 5V (red), GND (black) and GPIO4/D2 (yellow).
I've tryied with and without a pullup resistor (4k7 ohm) between 5V and GPIO port with no success.

Thanx in advance.
Danny

Danny89530, 2018-09-02:
Thanx @Valcob,
I've changed both VCC and sensor (just to be sure I've not burned it) but nothing changed.
I've also noticed that the issue was added to a milestone: can I consider my problem as a bug?

ColinShorts, 2018-09-02:
It might be worth trying a simple Arduino sketch to make sure that the board is seeing the sensor ok, I've had some fun trying to get a pair of ds18b20's working for tool for balancing radiators. There are sketches that will report the address of the sensor, and if they don't work you might need to check the polarity is correct (if it comes in a metal tube with a cable, some people have reported positive/negative swap). You could try blinking an led to make sure the gpio is still ok.

xoseperez, 2018-09-04:
The DS18B20 sensor was working last time I checked it myself. But it's a tricky one.

Misiu, 2018-09-05:
@Danny89530 DS18B20 should work fine with 5V and 4k7 resistor.
I'm building a small project using a couple of DS18B20 and Wemos board and everything works fine.
Try Your temperature sensors on separate board using Arduino IDE, when You install Dallas OneWire library You'll find a couple of examples in examples menu in Arduino IDE. There is one that list all DSXX sensors and reads the temperature.
There is an option that You ordered DS18B20 sensors from China and You got fake/broken sensors - https://blog.peku33.net/aliexpress-ebay-china-market-fake-ds18b20-sensors/

Danny89530, 2018-09-05:
Thanx for the suggestions but, just to be clear, the sensor was detached from a RaspPi and connected to the new board.
The sensor has worked for many years without any problem.
I will try with another Lulino board with a new sketch as suggestion from you but, with 3 different board and other 3 sensor, I've had the same problem with ESPurna.
Oh, I've tried to change also the pin, not just the D2, nothing changed.

Is there a way to "debug" it from ESPurna?

ColinShorts, 2018-09-05:
It'd be quicker making sure it works under Arduino. If it works under Arduino then it'll be easier to track down an issue in Espurna. It could be as simple as pin assignment in the lolin not being quite right.

xoseperez, 2018-09-06:
@Danny89530 The Dallas sensor code does not provide much debug info. You can check the boot log section where the device tries to find the DS18B20.

xoseperez, 2018-09-07:
I just tested the Dallas sensor with my test board and it is working fine. It must be a hardware issue, a proper boot log should say (my sensor is in GPIO13):

```
(...)
[000597] [SENSOR] Initializing Dallas @ GPIO13
[000598] [GPIO] GPIO13 locked
[000617] [SENSOR]  -> temperature:0
(...)
```

kakaki, 2018-09-08:
I'm preparing temperature sensor for each room in my house, already have made 3 boards with dallas sensor, all working ok - sensor is attached to pin 14 which is connected to 3.3V with pull up resistor 4,7k - tested on 1.13.1.

[image]

In config setting: 
		#ifndef DALLAS_SUPPORT
		#define DALLAS_SUPPORT             	1
		#endif
		#define DALLAS_PIN                 	14
		#define DALLAS_UPDATE_INTERVAL     	5000
		#define TEMPERATURE_MIN_CHANGE      1.0

Danny89530, 2018-09-10:
I was out this week end.
I will try this evening again.
Thanx for your feedback

Misiu, 2018-09-12:
@kakaki what modules are You using, it is ESP8266, but which board? Also, do You have links to DS18B20 sensors You use? I've ordered a couple from China, but none of them works 😞 (I know how to connect them, I have two old working fine)

Valcob, 2018-09-15:
@Misiu, as a board use the one you like the most. I like D1 mini  that's been my working horse for months now. You can get it here: [D1 mini ALIEXPRESS](http://s.click.aliexpress.com/e/bWABvugo)  or [D1 mini ALIEXPRESS 2](http://s.click.aliexpress.com/e/kxhpWyg) what stands for sensors well you never know what you'll get cause even from the same store you can easily get some refurb or just plain non working crap. But there is one small workaround to this you can get your money back ;) with a simple dispute.
For DALLAS sensors you can check these: [10 pcs DS18B20 ALIEXPRESS](http://s.click.aliexpress.com/e/4RN6rR6), [10 pcs DS18B20 ALIEXPRESS 2](http://s.click.aliexpress.com/e/cPdu5D0k), [10 pcs DS18B20 ALIEXPRESS 3](http://s.click.aliexpress.com/e/28hTxSC)
One small advice here get always couple or more (e.g 10) sensors :) cause you have to have some spare parts to burn first ))) on the way of your findings.

Hope this will help you.

Misiu, 2018-09-17:
@Valcob thanks for the links :) especially to ESP-M3. I already use a couple of D1 mini's, but I want something smaller for sensors.

Danny89530, 2018-09-19:
Ok, finally I've had the time to make a test and solve my issue.
I've made tests on 4 different boards and 4 different sensors and everytime I'd same result.
I've downloaded the example sketch of the Dallas library, powered the sensor with +3.3V, and a Pullup resistor and, once uploaded, everything worked at first try (this with the default sketch defined pin: D4/GPIO2).
Since I'm lazy, I've not changed the GPIO on the test board but I've modified the configuration on the platformio.ini precedently posted file and finally it worked.
Afther this result I've tested the other pins, like GPIO4, 14..but the sensor were not recognized..

Thanx for the support

closed 2018-09-19 by xoseperez