xoseperez/espurna #1179: 1.13.2 - Changed libs?

skyynet, 2018-09-02:
Sorry, but from 1.13.1 to 1.13.2 some to me unknown libs changed. Could you please indicate which?
I have ArduinoJson 6.3.0-beta installed which was working (needed?) in the last release. Do I have to revert to v5.something?

My error log while compiling:

Arduino: 1.8.3 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, DIO, 1M (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/Users/ingo/espurna/code/espurna/api.ino: In function 'void _onAPIs(AsyncWebServerRequest*)':
api:85: error: 'DynamicJsonBuffer' was not declared in this scope
         DynamicJsonBuffer jsonBuffer;
         ^
/Users/ingo/espurna/code/espurna/api.ino:85:9: note: suggested alternative:
In file included from /Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/DynamicJsonDocument.hpp:10:0,
                 from /Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:9,
                 from /Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from sketch/config/prototypes.h:2,
                 from sketch/config/all.h:33,
                 from /Users/ingo/espurna/code/espurna/espurna.ino:22:
/Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Memory/DynamicJsonBuffer.hpp:159:5: note:   'ArduinoJson::Internals::DynamicJsonBuffer'
     DynamicJsonBuffer;
     ^
api:85: error: expected ';' before 'jsonBuffer'
         DynamicJsonBuffer jsonBuffer;
                           ^
api:86: error: 'jsonBuffer' was not declared in this scope
         JsonObject& root = jsonBuffer.createObject();
                            ^
api:91: error: 'class ArduinoJson::JsonObject' has no member named 'printTo'
         root.printTo(output);
              ^
/Users/ingo/espurna/code/espurna/debug.ino: In function 'void _debugSend(char*)':
debug:71: error: 'DynamicJsonBuffer' was not declared in this scope
             DynamicJsonBuffer jsonBuffer(JSON_OBJECT_SIZE(1) + strlen(message) + 17);
             ^
/Users/ingo/espurna/code/espurna/debug.ino:71:13: note: suggested alternative:
In file included from /Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/DynamicJsonDocument.hpp:10:0,
                 from /Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:9,
                 from /Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from sketch/config/prototypes.h:2,
                 from sketch/config/all.h:33,
                 from /Users/ingo/espurna/code/espurna/espurna.ino:22:
/Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Memory/DynamicJsonBuffer.hpp:159:5: note:   'ArduinoJson::Internals::DynamicJsonBuffer'
     DynamicJsonBuffer;
     ^
debug:71: error: expected ';' before 'jsonBuffer'
             DynamicJsonBuffer jsonBuffer(JSON_OBJECT_SIZE(1) + strlen(message) + 17);
                               ^
debug:72: error: 'jsonBuffer' was not declared in this scope
             JsonObject &root = jsonBuffer.createObject();
                                ^
debug:81: error: 'class ArduinoJson::JsonObject' has no member named 'printTo'
             root.printTo(out);
                  ^
/Users/ingo/espurna/code/espurna/domoticz.ino: In function 'void _domoticzMqtt(unsigned int, const char*, const char*)':
domoticz:61: error: 'DynamicJsonBuffer' was not declared in this scope
             DynamicJsonBuffer jsonBuffer;
             ^
/Users/ingo/espurna/code/espurna/domoticz.ino:61:13: note: suggested alternative:
In file included from /Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/DynamicJsonDocument.hpp:10:0,
                 from /Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:9,
                 from /Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from sketch/config/prototypes.h:2,
                 from sketch/config/all.h:33,
                 from /Users/ingo/espurna/code/espurna/espurna.ino:22:
/Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Memory/DynamicJsonBuffer.hpp:159:5: note:   'ArduinoJson::Internals::DynamicJsonBuffer'
     DynamicJsonBuffer;
     ^
domoticz:61: error: expected ';' before 'jsonBuffer'
             DynamicJsonBuffer jsonBuffer;
                               ^
[... 596 lines cut ...]
/Users/ingo/espurna/code/espurna/ws.ino:463:5: note: suggested alternative:
In file included from /Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/DynamicJsonDocument.hpp:10:0,
                 from /Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:9,
                 from /Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from sketch/config/prototypes.h:2,
                 from sketch/config/all.h:33,
                 from /Users/ingo/espurna/code/espurna/espurna.ino:22:
/Users/ingo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Memory/DynamicJsonBuffer.hpp:159:5: note:   'ArduinoJson::Internals::DynamicJsonBuffer'
     DynamicJsonBuffer;
     ^
ws:463: error: expected ';' before 'jsonBuffer'
     DynamicJsonBuffer jsonBuffer;
                       ^
ws:464: error: 'jsonBuffer' was not declared in this scope
     JsonObject& root = jsonBuffer.createObject();
                        ^
ws:467: error: 'class ArduinoJson::JsonObject' has no member named 'printTo'
     root.printTo(output);
          ^
exit status 1
'DynamicJsonBuffer' was not declared in this scope

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

mcspr, 2018-09-02:
@skyynet yes and no, see https://github.com/xoseperez/espurna/wiki/ArduinoIDE / #924
> ESPurna is not yet compatible with ArduinoJson 6.X.X, use 5.13.2 instead

ArduinoIDE is not helping by showing -beta as first choice in library manager 🌵
some info: https://arduinojson.org/v6/doc/upgrade/?utm_source=github&utm_medium=release

closed 2018-09-07 by xoseperez

skyynet, 2018-09-07:
Thanks for the hint with ArduinoJson. This helped. Unfortunately I still have issues with NTPClientLib where 2.5.1 is installed.

ntp:48: error: 'class NTPClient' has no member named 'setNTPTimeout'
     NTP.setNTPTimeout(NTP_TIMEOUT);
         ^
/Users/ingo/espurna/code/espurna/ntp.ino: In function 'void _ntpConfigure()':
ntp:79: error: 'class NTPClient' has no member named 'setDSTZone'
     NTP.setDSTZone(dst_region);
         ^
exit status 1
'class NTPClient' has no member named 'setNTPTimeout'

skyynet, 2018-09-07:
Now I installed the development branch of NTPClientLib instead of the master branch and it compiles. Every update of ESPurna I have trouble with the libs. Hope it's just me...

mcspr, 2018-09-07:
@skyynet consider using [platformio](http://docs.platformio.org/en/latest/installation.html#macos-homebrew), libs will be downloaded/updated automatically with correct versions.