xoseperez/espurna #1183: Home Assistant configuration contains invalid name

robin13, 2018-09-06:
espurna has a limit on hostname that a hostname cannot contain and underscore `_` (but does allow a minus (`-`)), while home assistant has a limit that an entity may only contain alphanumeric plus underscore for the name.

If you name your espurna device e.g. `sonoff-th16-01` this will result in a configuration:

```
switch:
  - name: sonoff-th16-01    <--- invalid (a)
    platform: mqtt
...

sensor:
  - name: sonoff-th16-01 temperature   <- invalid (b)
```

(a) is invalid because it has `-` used, and (b) is invalid because there are both `-` and spaces (` `) used.
It would be good to check (I think around [here](https://github.com/xoseperez/espurna/blob/dd12251e43174281c737ca0868d5e6fcb8711558/code/espurna/homeassistant.ino#L172)) if the resulting name is valid by home assistant entity standards (ascii alphanumeric + underscore) and fix if necessary.

Valcob, 2018-09-06:
@robin13, you can use custom names for switches so you can set it sonoff_th16_01 and it should work the same way as before. I have all my devices set up this way not a problem like at all. Please try and report back if you have any issues or questions.

robin13, 2018-09-06:
@Valcob This is true, but
1. setting a custom name is one additional manual step to do when setting up your device - less manual customisation will result in less mistakes
1. whether you define a custom name or not (and whether you choose to put characters in there that are valid for home assistant entities or not), the resulting home assistant configuration should be valid for home assistant
1. the automatic name generated for sensor are always invalid because they contain a space.

xoseperez, 2018-09-06:
Hostnames follow the rules for hostnames. I think the best option would be to rework names for HA removing non-valid characters.

Valcob, 2018-09-06:
@xoseperez do we need that extra space in name for magnitudes?
`_haFixName(getSetting("hostname") + **String(" ")** + magnitudeTopic(type));`

xoseperez, 2018-09-06:
@Valcob It's only to make it look good: imagine you have an outdoors device named "outdoors" with a temperature sensor, the HA name would be "outdoors_temperature". Looks nicer than "outdoorstemperature"

Valcob, 2018-09-06:
sorry didn't noticed the closing bracket for _haFixName :) for some reason I saw it as only applying for getSetting("hostname"). Sorry my bad )

xoseperez, 2018-09-06:
Will be released with next version

closed 2018-09-06 by xoseperez