xoseperez/espurna #1186: Generic digital input bug in mqtt?

kroon040, 2018-09-09:
Hi,

I compiled espurna on gpio4 and looks like something goes wrong in mqtt.

When the gpio pin goes up or down, you see it in the web interface status , but looking in de debug, no mqtt message is send, the mqtt messages comes , but at pre defined updates I think.

It should when it change the mqtt wil also be send immediate, like the web interface status does

Is this as designed or a bug?

xoseperez, 2018-09-10:
When you say "espurna on gpio4" what do you mean? Are you using the EventSensor?

kroon040, 2018-09-10:
Sorry,

I compiled digital pin op gpio4

    #ifndef DIGITAL_SUPPORT
    #define DIGITAL_SUPPORT      1
    #endif
    #define DIGITAL_PIN          4
    #define DIGITAL_PIN_MODE     INPUT

I'm using it as detection when the garage door opens (1 is closed, 0 is open)

[724689] [MQTT] Sending Garagedeur/digital => 1 (PID 1)

The web interface the value change rapidly, but de mqtt message comes later.

I first tried it with button switch, but that only sends the value 2.

So now using the digital sensor, but that looks like it is reacting slow in mqtt

I solved it a little in the web interface , to set read interval in 6 seconds and send report every 1. Than i get every 6 second and mqtt event, but that is not the best solution I think.

Richard

kroon040, 2018-09-10:
ok, i'm trying, I think I need than the events_interrupt_mode set to both? 

I have this in de hardware.h

    #define EVENTS_SUPPORT      1
    #define EVENTS_PIN          4
    #define EVENTS_TRIGGER      1
    #define EVENTS_INTERRUPT_MODE     BOTH

But I get an error that both is not declared in this scope

Compiling .pioenvs/nodemcu-lolin/src/espurna.ino.cpp.o
In file included from espurna/config/all.h:29:0,
from /home/richard/espurna/nieuw/espurna/code/espurna/espurna.ino:22:
/home/richard/espurna/nieuw/espurna/code/espurna/sensor.ino: In function 'void _sensorLoad()':
espurna/config/hardware.h:86:39: error: 'BOTH' was not declared in this scope
#define EVENTS_INTERRUPT_MODE     BOTH
^
/home/richard/espurna/nieuw/espurna/code/espurna/sensor.ino:469:34: note: in expansion of macro 'EVENTS_INTERRUPT_MODE'
sensor->setInterruptMode(EVENTS_INTERRUPT_MODE);
^

ColinShorts, 2018-09-10:
#define EVENTS_INTERRUPT_MODE           RISING  // RISING, FALLING, CHANGE

Maybe try CHANGE

kroon040, 2018-09-11:
I compiled, but the event sensor also triggers when the other gpio5 is triggered, my relais to open the garage. so it reacts very different, tried all options

The digital pin is much better and stable, only sends not the trigger. Is it possible to have digital_trigger 1??

xoseperez, 2018-09-11:
The EventSensor should only monitor the GPIO you specify.

theknut, 2018-11-13:
I'm havin issues with the EVENTS_SUPPORT. I want to use GPIO14 to detect when my doorbell is pressed. I use the following defines:
`#define SENSOR_SUPPORT				1`
`#define EVENTS_SUPPORT				1`
`#define EVENTS_SUPPORT				1`
`#define EVENTS_TRIGGER				1`
`#define EVENTS_PIN    				14`
`#define EVENTS_PIN_MODE      		INPUT`
`#define EVENTS_INTERRUPT_MODE		CHANGE`
`#define EVENTS_DEBOUNCE      		50`

However, whenever I put 3.3V on GPIO14 I don't get any recognition from espurna. What am I doing wrong?

@kroon040 You seem to have the same goal as I have. Did you figure it out eventually? Plus, why do you use GPIO4? I'm using a Sonoff Basic, which board are you using?

closed 2019-01-20 by stale[bot]