xoseperez/espurna #1192: REST API changes values on GET

Cabalist, 2018-09-11:
Currently the RESTFUL API will respond to GETs as if they were PUTs.    For example:

This python code toggles the relay on my device:

```python
    r = requests.get(f"http://{v['ip']}/api/relay/0", params={'apikey': v['apikey'], 'value': 2})
```

This should not happen as a GET request is 'supposed' to be read only and a [Safe Method](https://en.wikipedia.org/wiki/Hypertext_Transfer_Protocol#Safe_methods)

Tested on 1.13.2 with a Sonoff B1.

xoseperez, 2018-09-13:
I agree this breaks the REASTFUL API. It was requested to easy access from certain clients. Maybe it should be disabled by default..

xoseperez, 2018-09-25:
Changed the default behaviour so now the API is restful "compliant". It can be changed from the web UI.

closed 2018-10-05 by xoseperez