xoseperez/espurna #1193: HASS show configuration: # recognized as comment in yaml file

Destix, 2018-09-12:
ESPURNA 1.13.3a

HASS > click on Show button 

if there are a few switches 

```  
- name: garage-gate #0 
``` 
equals  

``` 
- name: garage-gate #1 
``` 
in HASS config because # recognized as comment in yaml file

Example
```
switch:
  - name: garage-gate #0
    platform: mqtt
    state_topic: garage-gate/relay/0
    command_topic: garage-gate/relay/0/set
    payload_on: 1
    payload_off: 0
    availability_topic: garage-gate/status
    payload_available: 1
    payload_not_available: 0

switch:
  - name: garage-gate #1
    platform: mqtt
    state_topic: garage-gate/relay/1
    command_topic: garage-gate/relay/1/set
    payload_on: 1
    payload_off: 0
    availability_topic: garage-gate/status
    payload_available: 1
    payload_not_available: 0
```

xoseperez, 2018-09-16:
Commit 10ec1b20390ac69931e5603a1219bb421f2b280e should fix this as well, any non alphanum char is translated into a "_" for the device name...

xoseperez, 2018-10-07:
This is a generated YAML after the 10ec1b2 commit:

```
switch:
  - name: ESPURNA_0DD36A_0
    platform: mqtt
    state_topic: ESPURNA-0DD36A/relay/0
    command_topic: ESPURNA-0DD36A/relay/0/set
    payload_on: 1
    payload_off: 0
    availability_topic: ESPURNA-0DD36A/status
    payload_available: 1
    payload_not_available: 0

switch:
  - name: ESPURNA_0DD36A_1
    platform: mqtt
    state_topic: ESPURNA-0DD36A/relay/1
    command_topic: ESPURNA-0DD36A/relay/1/set
    payload_on: 1
    payload_off: 0
    availability_topic: ESPURNA-0DD36A/status
    payload_available: 1
    payload_not_available: 0
```

As you can see all non-alphanum chars except '_' are replaced by the '_' char.

closed 2018-10-07 by xoseperez