xoseperez/espurna #1194: Extra GPIO on t1?

ways, 2018-09-12:
Here are some new pictures for you of the latest T1, use as you wish.

[image]
[image]

I see there's a port by the normal gnd, tx, rx, vcc ports marked "LOG". Do you know if that is an extra GPIO which can be used?

ways, 2018-09-12:
If not, can the GPIO meant for the middle button be used? How?

mcspr, 2018-09-12:
Isn't it the same as previous version? Being GPIO2
Tasmota wiki states that it is the same https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU#eu-model-of-1-button-t1-received-direct-from-sonoff-china-2018-06-04

ways, 2018-11-12:
An update before I close this; I tried using "log", but the power needed for a PIR sensor made the board crash all the time, so I gave up.

closed 2018-11-12 by ways

Valcob, 2018-11-12:
@ways What type of PIR did you use ? how did you connect it to the board? did you measure the consumption of the PIR and voltage drop with it on the AMS?

ways, 2018-11-12:
> @ways What type of PIR did you use ? how did you connect it to the board? did you measure the consumption of the PIR and voltage drop with it on the AMS?

A common PIR module from dx.com or similar. Also tried a small varian (from ebay?). I didn't do much measurements. If you think it's supposed to work, I'll do a proper test, and create a new issue (or wiki documentation) with the results.