xoseperez/espurna #1196: HEARTBEAT_ENABLED is overwritten despite what the docs says

wsw70, 2018-09-12:
I flashed a new version of ESPURNA to my devices after heavily customizing it via `custom.h`. One of the changes in the new firmware was to disable the heartbeat by setting in `custom.h`

    #define HEARTBEAT_ENABLED 0

**Upon connecting to MQTT, the device sent a heartbeat**. It did not send anymore, as far as I can tell after watching the MQTT console for >5+ minutes (30 min actually, so there would have been ample time to send it a few times already).

**Is this a one-shot message which is sent upon connection?**

Now, the [documentation](https://github.com/xoseperez/espurna/wiki/Software-features#overview-of-the-feature-groups) states that 

> `*_ENABLED` (...) Changing this flag will set the default behavior of a new device, but will not override the saved configuration of old device.

I understand this as _"if `HEARTBEAT_ENABLED` was set to "something", flashing a new firmware will not modify this "something""_. This is not very clear as the parameter must have been set a way or another in the past. 

**The current behaviour (if the heartbeat was a one-time message) is perfect.** I am just mentioning that the documentation is not exactly clear on that point.

mcspr, 2018-09-19:
Hello.

You are right, this probably should be HEARTBEAT_SUPPORT instead.

And current code strips out heartbeat completely when setting HEARTBEAT_ENABLED to 0, so you probably seeing retained ones. By default mqtt client sends retain flag so latest topic/message are saved by the broker and you will see them when subscribing. ESPurna does not clear retained messages, it is up to the broker or some other client app to clear them.

wsw70, 2018-09-20:
> so you probably seeing retained ones

I will check this out tonight but I do not think so. I saw the messages (once) right at the moment when the device connected to MQTT. 
I was monitoring the bus before (so I saw all the retained messages pop-up when I connected my monitoring client)

xoseperez, 2018-09-24:
The documentation is wrong and it should probably be named HEARTBEAT_SUPPORT as @mcspr  says. But what would be the expected behavior? The heartbeat is something that happens at regular intervals (5 minutes by default). 

Currently, there is a call to systemSendHeartbeat() from the main MQTT connect event handler, but that call will not do anything if HEARTBEAT_ENABLED is 0. Maybe we should be able to configure separately the initial heartbeat (to say "hey, I'm here") form the regular one.

wsw70, 2018-09-24:
I am not sure the name is wrong. My point was that there is an initial heartbeat sent (a bunch of status messages, including all relay status) even when the `HEARTBEAT_ENABLED` is `0`. I believe that this initial message should not be sent out either (there is no particular reason to send it, I think).

If you wonder why I had this problem at all - it was because the heartbeat was published on MQTT, including the delay status. My approach is that there is a message for the relays only when something changed (that was a trigger in my case). Disabling the heartbeat fixed the problem (and that initial heartbeat I can live with - it is just that the docs state that it is a on:off switch, without mentioning this initial message)

xoseperez, 2018-09-24:
I cannot reproduce this issue. If I set HEARTBEAT_ENABLED to 0 in my `custom.h` file and build the firmware using -DUSE_CUSTOM_H there is no heartbeat message. Are you sure your `custom.h` file is being loaded?

wsw70, 2018-09-24:
@xoseperez you mean that the start of the device does not issue a heartbeat message (a set of messages, actually)?

I do use `-DUSE_CUSTOM_H` when compiling (there are several changes I make to the firmware this way) and I will test this once I am at home later this week. Thanks.

xoseperez, 2018-09-25:
The set of messages is produced by the `heartbeat()` method in `utils.ino`, and, as you can see it's only available if HEARTBEAT_ENABLED is 1...

xoseperez, 2018-09-26:
I've just added code to be able to configure the heartbeat in 3 different scenarios depending on the value of a HEARTBEAT_MODE setting:

- HEARTBEAT_NONE: no heartbeat at all, never
- HEARTBEAT_ONCE: only on MQTT connection
- HEARTBEAT_REPEAT: on MQTT connection and after HEARTBEAT_INTERVAL

The olf HEARTBEAT_ENABLED is mapped to HEARTBEAT_NONE if 0 or HEARTBEAT_REPEAT if 1. HEARTBEAT_REPEAT is the default value for backwards compatibility.

wsw70, 2018-09-27:
@xoseperez : thanks, this is wonderful.

closed 2018-09-27 by wsw70