xoseperez/espurna #1200: MQTT Message to Turn Of Switch On Rf Bridge

seetendra, 2018-09-15:
Hello Everyone 

I am trying to get a MQTT Switch to Turn On an Switch on the Rf Bridge and then an automation to automatically turn off the Switch with in 5 seconds
 As of now i have been able to turn on the Switch but for some reason iam not able to turn it off.
Please do let me know what am i doing wrong 

- alias: BellRinging
  hide_entity: True
  trigger:
    platform: mqtt
    topic: RFBridge/relay/0
    payload: '1'
  action:
    - service: xiaomi_aqara.play_ringtone
      data:
        gw_mac: f0b429xxxxx
        ringtone_id: 10
        ringtone_vol: 10
    - delay: '00:00:10'
    - service: mqtt.publish
      data:
        topic: RFBridge/relay/0
        payload: '0

mcspr, 2018-09-16:
Hello.
action mqtt topic should have /set, check out the wiki: https://github.com/xoseperez/espurna/wiki/MQTT#commands-and-states

seetendra, 2018-09-20:
That worked like a charm. You guys are the best.

closed 2018-09-20 by seetendra