xoseperez/espurna #1205: Build project error - utils.ino:23:5: error: expected unqualified-id before '||' token

icevoodoo, 2018-09-17:
Build project error see below:

Processing itead-sonoff-th (framework: arduino; platform: espressif8266@1.5.0; board: esp01_1m)
-------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
SYSTEM: ESP8266 80MHz 80KB RAM (1MB Flash)
Converting espurna.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 51 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 5.13.2
|-- <AsyncMqttClient> 0.8.1 #ddbf4d1
|   |-- <ESPAsyncTCP> 1.1.3 #55cd520
|   |   |-- <ESP8266WiFi> 1.0
|-- <Brzo I2C> 1.3.2
|-- <DebounceEvent> 2.0.4 #4681547
|-- <EEPROM_Rotate> 0.9.1 #9c7d14f
|   |-- <EEPROM> 1.0
|-- <Embedis> 1.2.0
|-- <Encoder>
|-- <EspSoftwareSerial> 3.4.1 #5378868
|-- <ESPAsyncTCP> 1.1.3 #55cd520
|   |-- <ESP8266WiFi> 1.0
|-- <ESP Async WebServer> 1.1.0 #05306e4
|   |-- <Hash> 1.0
|   |-- <ESPAsyncTCP> 1.1.3 #55cd520
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 5.13.2
|   |-- <ESP8266WiFi> 1.0
|-- <FauxmoESP> 3.0.1 #9a03a61
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.1.3 #55cd520
|   |   |-- <ESP8266WiFi> 1.0
|-- <HLW8012> 1.1.0 #cc3baec
|-- <IRremoteESP8266> 2.2.0 #4d57a2b
|-- <JustWifi> 2.0.2 #1367a8b
|   |-- <ESP8266WiFi> 1.0
|-- <mDNSResolver> 0.1 #4cfcda1
|   |-- <ESP8266WiFi> 1.0
|-- <my92xx> 3.0.1 #cda124a
|-- <NoFUSS> 0.2.5 #1835206
|   |-- <ArduinoJson> 5.13.2
|   |-- <ESP8266httpUpdate> 1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266HTTPClient> 1.1
|   |   |   |-- <ESP8266WiFi> 1.0
|-- <NtpClientLib> 2.5.0 #0942ebc
|   |-- <Time> 1.5 #ecb2bb1
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <OneWire> 2.3.4
|-- <PZEM004T> 1.1.2
|   |-- <EspSoftwareSerial> 3.4.1 #5378868
|-- <PubSubClient> 2.6
|-- <rc-switch> 2.6.2
|-- <RFM69> #cb68f71
|   |-- <SPIFlash>
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|-- <Time> 1.5 #ecb2bb1
|-- <NewPing> 1.9.1
|-- <Ticker> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0
|-- <ESP8266SSDP>
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs\itead-sonoff-th\src\espurna.ino.cpp.o
Compiling .pioenvs\itead-sonoff-th\src\fs_math.c.o
Compiling .pioenvs\itead-sonoff-th\src\pwm.c.o
Archiving .pioenvs\itead-sonoff-th\libb18\libArduinoJson_ID64.a
Compiling .pioenvs\itead-sonoff-th\lib537\ESP8266WiFi\ESP8266WiFi.cpp.o
Compiling .pioenvs\itead-sonoff-th\lib537\ESP8266WiFi\ESP8266WiFiAP.cpp.o
Compiling .pioenvs\itead-sonoff-th\lib537\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
Compiling .pioenvs\itead-sonoff-th\lib537\ESP8266WiFi\ESP8266WiFiMulti.cpp.o
Compiling .pioenvs\itead-sonoff-th\lib537\ESP8266WiFi\ESP8266WiFiSTA.cpp.o
D:/GitHub/xoseperez/code/espurna/utils.ino:23:5: error: expected unqualified-id before '||' token
#else
^
Compiling .pioenvs\itead-sonoff-th\lib537\ESP8266WiFi\ESP8266WiFiScan.cpp.o
Compiling .pioenvs\itead-sonoff-th\lib537\ESP8266WiFi\WiFiClient.cpp.o
*** [.pioenvs\itead-sonoff-th\src\espurna.ino.cpp.o] Error 1
================================================= [ERROR] Took 22.15 seconds =================================================

mcspr, 2018-09-17:
I can reproduce this when `code/espurna/utils.ino` has CRLF line terminators (thanks dos2unix), for some reason backslash at the end of the line triggers this 😕

mcspr, 2018-09-17:
@icevoodoo can you try with changes from #1208? .ino -> .h switch seems to help.

icevoodoo, 2018-09-20:
It's ok now, after today updates... 10x!

closed 2018-09-20 by icevoodoo