xoseperez/espurna #1207: MQTT group - Without Broker

jpedrocarvalho, 2018-09-17:
Hi,
I have 2 boards falshed with ESPURNA. I would like to know if it is possible to use MQTT group between the  2 withou a broker.
The idea is if the relay changes on one board, it should sync and change it on the other board.

Is this possible?

Thanks
JP

Valcob, 2018-09-18:
@jpedrocarvalho, The MQTT broker is the source of your desired synchronization. So without a subscription to a valid MOTT broker there is no way to broadcast a state change to other devices on your network or to the outside world. Long story short on how MQTT broker works - you devices are subscribing to some custom topics and then after a successful subscription they can publish their state and listen for a state change from others, this way the communication  between devices is possible (more on this [Why do we need MQTT](https://www.youtube.com/watch?v=exMm-fmU5ck&feature=youtu.be&t=27))  you can use some sort of Thingspeak and back web API calls in combination with some IFTT service but that would involve so much effort and so many security concerns so would suggest you better not go this way.

Hope this responds your question.

jpedrocarvalho, 2018-09-18:
@Valcob, Thank you very much for your reply.
I've been reading what I can find online and I though that mite be the case.
I just think that simple situations like mine, where I just want to have 2-3 ESP8266 synced between them, so I can do a sort of virtual 3-WAY switch, having to use a dedicated PC or Raspberry for that is a little to much.
And even for redundancy, since if for some reason the broker (pc) goes down, I can continue still use the switches.

For what I've been able to find online it shouldn't be that difficult to broadcast a message from one ESP to all others with something like GROUP_ID/SWITCH#, to that another ESP that is configured with that GROUP_ID toggled the same SWITCH (maybe a good future addition to ESPURNA.....)
But since it isn't developed I can either try do do it myself or go done the dedicated broker root.

Thanks anyway.

JP

ColinShorts, 2018-09-18:
Why not run a broker on an esp8266 then? The are a few Arduino libraries for it, eg: https://github.com/martin-ger/esp_mqtt/blob/master/README.md

jpedrocarvalho, 2018-09-18:
Hi @ColinShorts ,
Thanks for the reply. I will try to use that library.
Since all my ESP are for light switching, I will try to include the broker in one of them. Hopefully it won't break the normal ESPURNA functionality.

Thanks
JP

closed 2019-01-25 by stale[bot]