xoseperez/espurna #1210: Dallas 1-Wire sensors

flashy02, 2018-09-19:
Hi, I ve a question. Which Dallas OW-Sensors are supported? Only DS18B20 or others like DS2438 with p.e. additional humidity-sensor too? Or, is it possible to use the sensors offered here: https://www.tm3d.de/shop/kategorien/module

Regard Michael

ColinShorts, 2018-09-19:
Hi flashy02, it looks like that the DS2438 is supported by the OneWire library - at least according to [Ardunio playground](https://playground.arduino.cc/Learning/OneWire), however it doesn't look like Espurna supports it (yet?), not even in the sensors branch.

Have a look at the files in [code/espurna/sensors](https://github.com/xoseperez/espurna/tree/dev/code/espurna/sensors) for currently supported sensors if the list in the [wiki](https://github.com/xoseperez/espurna/wiki/Sensors) isn't sufficient.

flashy02, 2018-09-20:
Hi Colin, thank you for the fast response and the hint to the "DallasSensor.h". An additional enhancement would be the DS2450 like here https://www.tm3d.de/elektronik-projekte/1-wire-sim-beispiele/ds2450/ds2450-multisensor as multisensor. With that, it is possible to use one OneWire-sensor to get e.g. temperature, humidity, illumination and air pressure