xoseperez/espurna #1212: No web interface after a fresh install

AlbrtoSam, 2018-09-19:
I had installed espurna-1.12.3-espurna-core.bin on my dual sonoff, and now i cant go to web interface. 

I can connect using telnet. I dont want to flash opening device. Once I already access via telnet, is there a way to flash a complete version with web interface enabled ?

m[681672]

          [681673] [INIT] ESPURNA 1.12.3
                                        [681673] [INIT] xose.perez@gmail.com
                                                                            [681673] [INIT] http://tinkerman.cat

                                                                                                                [681673] [INIT] CPU chip ID: 0x86C20B
                                                                                                                                                     [681674] [INIT] CPU frequency: 80 MHz
                                                                                                                                                                                          [681678] [INIT] SDK version: 1.5.3(aec24ac9)
                                                                                                                                                                                                                                      [681682] [INIT] Core version: 2_3_0
                            [681685] [INIT] Core revision:
                                                           [681688]
                                                                    [681689] [INIT] Flash chip ID: 0x144051
                                                                                                           [681692] [INIT] Flash speed: 40000000 Hz
                                                                                                                                                   [681695] [INIT] Flash mode: DOUT
                                                                                                                                                                                   [681698]
                                                                                                                                                                                            [681699] [INIT] Flash sector size:     4096 bytes
                                                                                                                                                                                                                                            [681704] [INIT] Flash size (CHIP):  1048576 bytes
                                                [681708] [INIT] Flash size (SDK):   1048576 bytes /  256 sectors
                                                                                                                [681714] [INIT] Firmware size:       296480 bytes /   73 sectors
                                                                                                                                                                                [681719] [INIT] OTA size:            729088 bytes /  178 sectors
   [681725] [INIT] SPIFFS size:              0 bytes /    0 sectors
                                                                   [681730] [INIT] EEPROM size:           4096 bytes /    1 sectors
                                                                                                                                   [681736] [INIT] Empty space:          16384 bytes /    4 sectors
                                                                                                                                                                                                   [681742]
                                                                                                                                                                                                            [681743] [INIT] BOARD: ITEAD_SONOFF_BASIC
        [681746] [INIT] SUPPORT: BROKER DEBUG_SERIAL DEBUG_TELNET MDNS_SERVER TELNET TERMINAL THINGSPEAK

                                                                                                        [681755] [INIT] Last reset reason: Software/System restart
                                                                                                                                                                  [681760] [INIT] Settings size: 180 bytes
                                                                                                                                                                                                          [681764] [INIT] Free heap: 31752 bytes
   [681767] [INIT] Power: 3186 mV
                                 [681769] [INIT] Power saving delay value: 10 ms
                                                                                [681774]
                                                                                         [681775] [WIFI] MODE AP --------------------------------------
                                                                                                                                                       [681780] [WIFI] SSID  ESPURNA_86C20B
                                                                                                                                                                                           [681783] [WIFI] PASS  fibonacci
                                                                                                                                                                                                                          [681786] [WIFI] IP    192.168.4.1
              [681790] [WIFI] MAC   DE:4F:22:86:C2:0B
                                                     [681792] [WIFI] ----------------------------------------------
                                                                                                                   [681798] +OK

ColinShorts, 2018-09-19:
You should hopefully be able to flash it using platformio;

$ pio run -e itead-sonoff-dual-ota -t upload

AlbrtoSam, 2018-09-20:
I finally could recover using python script. Thank you!.

closed 2018-09-24 by xoseperez