xoseperez/espurna #1213: Not reporting Switch status 0 to ThingSpeak

JohnOH, 2018-09-20:
When ThingSpeak is enabled the value of Switch (on) is reported to ThingSpeak correctly as a value 1, however Switch (off) is not reported when it should be reported as a value 0.

JohnOH, 2018-09-20:
Unit is now reporting Switch 0 to ThinkSpeak. I may have reported this as an issue to early or there is some initialisation issue.

xoseperez, 2018-09-24:
I have sometimes seen the message fail with a return payload of "0" instead of the current dataset index. I have added code to the thingspeak module to try up to 3 times to send the message if the return code is not right.

xoseperez, 2018-09-25:
The POST is done to the `update` URL also (the last update clarifies that too), the only difference is the api key parameter. In the documentation they state it must be `api_key` (https://www.mathworks.com/help/thingspeak/writedata.html)

closed 2018-09-25 by xoseperez