xoseperez/espurna #1214: Fatory reset on config change

DJBenson, 2018-09-20:
**Device**: Sonoff TH16
**Build**: 1.13.3a
**SDK**: 2.2.1(cfd48f3)

I have a Sonoff TH16 which I'm using to measure the temperature in an outbuilding containing some animals, if the temperature dips below a threshold, it turns on the switch (which is connected to a 2500w heater). I use Domoticz to drive the logic and also have MQTT in the mix (although not necessarily used in this configuration).

I have noticed that when making certain config changes, the UI freezes and on refreshing the config has reverted to factory.

The past two times it has happened, it has been when I disabled the MQTT client.

Is there any way I can debug the issue?

DJBenson, 2018-09-20:
FYI I've just flashed 1.13.2 pre-built over the top and the same issue does not happen. Unless you can think of a specific reason why this would happen with my build, I'm not sure it's worth investigating - I need this device to be reliable so I'm probably going to use release builds on it rather than building my own.

mcspr, 2018-09-23:
Hello.

There is known issue when turning mqtt off, caused by library used here. See https://github.com/xoseperez/espurna/issues/1115

And besides possible bug between 1.13.2 and 1.13.3a, notable difference is releases are built with old Arduino Core 2.3.0 / SDK 1.5.3, while the most recent one is 2.4.2 / SDK 2.2.1

But the mqtt itself may not be related - I too recently had similar reset incident after changing settings on long running device, after ota update to 1.13.3a from 1.13.2, both with 2.4.2 core. Settings were not truly 'factory reset' though, admin password and wifi settings were still there, but not settings related to http api, mqtt client and led. And device itself did not reboot, uptime still being 2+ days since the ota.

mcspr, 2018-10-03:
Looking at 2.3.0...2.4.2 history for eeprom lib, it started handling _dirty flag slightly different:
https://github.com/esp8266/Arduino/commit/1b76e9bf72c05933ed912d2603bfa2d057db800e#diff-50c0132dfa7f533ff53573a3554778f0

@xoseperez might not be related... is it valid to call EEPROM_Rotate::begin twice? these called in order in setup():
https://github.com/xoseperez/espurna/blob/88e2eccccef3846845c2b43d5b58e3a24c61ef2d/code/espurna/eeprom.ino#L89
https://github.com/xoseperez/espurna/blob/5b62978d093089e99aba79405710556c08a9968c/code/espurna/settings.ino#L446

xoseperez, 2018-10-05:
Aside from the fact that it's completely useless to call it twice, I can't see a problem. But I have removed the call from the settings module.

mcspr, 2018-10-07:
Right, only thing it did is populate '_data' from the flash and set initial sector (as numbers were the same - same sector was chosen)

I did notice another thing though:
https://github.com/xoseperez/espurna/blob/acf68ac5c8dbf09473b359d94792dda51201913b/code/espurna/relay.ino#L177
https://github.com/xoseperez/espurna/blob/acf68ac5c8dbf09473b359d94792dda51201913b/code/espurna/relay.ino#L386-L396
May .commit() take a long time here i.e. should this be saveSettings() instead?
And maybe it should be guarded from writing flash unless 'boot_mode' is set to RELAY_BOOT_SAME / TOGGLE ?

xoseperez, 2018-10-07:
Yes, those two changes will mean an improvement, for sure. I'm not sure any of those are the reason for the factory reset thou.

mcspr, 2018-10-07:
Maybe shot in the dark. But I also did see couple of other 'commit()' calls that should be delayed until next loop, so i'll make a PR for that.

General question about eeprom storage. For 1MB layout, EEPROM_Rotate sets offset for crc to 0 and counter to 2. Looking at 'flash.dump' output:
```
flash.dump 250

         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
------------------------------------------------------
0x0000:  00 FF FF FF FF 00 00 FF FF FF FF 71 5F B9 FF FF

flash.dump 251

         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
------------------------------------------------------
0x0000:  00 FF FF FF FF 00 01 FF FF FF FF 71 60 B8 FF FF
```
Judging by couple of relay switches and seeing the values:
CRC is at `11` and `12`
Counter is at `13`

xoseperez, 2018-10-07:
Right. Sorry, I just pushed the changes along with a couple more commits, I guess they were the same you were talking about.

About EEPROM_Rotate, I added a method to the library to have the option to specify the offset to the CRC in the EEPROM since I knew ESPurna was using the first 11 bytes. This is the EEPROM_ROTATE_DATA setting and it is initialized in the eepromSetup method:

https://github.com/xoseperez/espurna/blob/6fee2f1f9fd6ff4fe0f921050a2d1c3404b319d0/code/espurna/eeprom.ino#L88

closed 2019-01-20 by xoseperez