xoseperez/espurna #1216: SonOFF Dual Switches wrong behavior

Bochets34, 2018-09-21:
Hello,

I'm using Espurna successfully with several SonOFF modules, and especially the Dual version to control Roller blinds.
my setup is to drive the two relays for rolling up & down from Domoticz through http request. fine, works great
I also use 'regular' switches' on the wall connected to the Dual SonOFF ports, working great too, but ...
the following scenario is my  issue:
1) Request from the Wall Switch to open the blind
2) dont put the switch back to the 'neutral' position, thus still one port has shortcut
3) using Wifi,through Domoticz or Web browser directly, ask to close the blind, working.
4) back to the wall trying to open the blind, works
5) trying to stop using the neutral (thus all switches off) : doesn't stop
6) it goes back to closing instead

workaround: cycle through all switches combination in order to get back to normal behavior: confusing for kids or old people

it looks like Espurna doesn't manage values on input ports the right way, if relays have been controlled  through Wifi, to me it doesn't look like an expected behavior: is it ?
probably when one switch was set to 1 before the Wifi command, its value is not cleared, unlike 'switch only mode' leading to this issue.

could you please look at this scenario and fix it?
I'll be glad to provide more details if needed, or help once I know where the piece of code is located , I'm a bit programmer myself :)

thanks
Olivier

xoseperez, 2018-09-25:
The Dual is a tricky device because ESPurna does not control the buttons or relays itself. Instead, there is a secondary microcontroller in charge of that. ESPurna does not get "release" events from that uC.

skyynet, 2018-10-08:
Just a hint: Upgrading to the Sonoff Dual R2 version should fix that as the additional microcontroler was removed. I just bricked every r2 module I tried so far :-(

Bochets34, 2018-10-09:
I'm already using Dual R2...over 10 total, same behavior

macobra, 2018-11-22:
My Dual R2 seems to have 'boot mode' for switch#1 reversed.
While switch#0 'boot mode'='Always ON/OFF' results in an ON/OFF according to the GUI - meaning is working as expected,
switch#1 'bootmode'='Always ON' starts with OFF after a restart and vice versa (starts OFF when set to ON).

Just adding here instead opening a new issue since both misbehaviours might have the same cause.

mcspr, 2018-11-28:
Could this be related to button setup?
https://github.com/xoseperez/espurna/blob/e2641b9aed5e50e2502022fb0c029c4fdef09252/code/espurna/config/hardware.h#L499-L501

I am not that familiar with pin initialization requirements, but reading DebounceEvent source  BUTTON_SWITCH initial reading is done without pinMode setting it INPUT_PULLUP (or INPUT at all):
https://github.com/xoseperez/debounceevent/blob/4681547f72b226a9bafc95cf7a4623ec69f6a327/src/DebounceEvent.cpp#L36-L65

Core itself skips this too
(2.3.0 version linked, same with 2.4.X. My assumption that is because generic ESP8266 connects GPIO9/10 to SPI flash, unlike ESP8285 that is used in Dual R2):
https://github.com/esp8266/Arduino/blob/09826c6d87e54c15422da90f060377c9ef188947/cores/esp8266/core_esp8266_wiring_digital.c#L160-L176

xoseperez, 2019-02-10:
Sorry, I'm back to this now. The Dual ports in the board are defined by default as BUTTON_SWITCH and that could be the issue since ESPurna does not have any means to know the status of the button other than that it flipped. Using other means to change the status of the relay (like the web interface) will be confusing since the status of the relay will not be sync'ed with the physical button. 

Can you report how do you have the different buttons wired. Excuse me I'm not into automated blinds but, what is the "neutral"?

Bochets34, 2019-02-11:
Actually I'm using the 'regular' double or triple switches.
triple is actually a double which mechanically disengage what ever position the switch is in (up or down)
double switch has also a mechanical means preventing from having both switches ON...(like the switch sync mode in the Web interface set to Zero or One switch active)
The electrical connection is simple: 'neutral' or common/shared is ground/0V, while each other point is connected to the buttons 0 and 1, as labelled on the Dual R2 PCB.

hope this helps
Olivier