xoseperez/espurna #1222: Quinled + HASS + 2 strips = only on/off switch

zen, 2018-09-22:
Hi,
I'm using  2 channel LED controller (Quinled) and with HASS autodiscovery it acts as a switch, only allowing to switch on / off both channels. Is there any way control it separately in HASS?

zen, 2018-09-25:
So, my temp workaround is not to use main switch, and use channels through a trick in HASS:

- platform: mqtt
  name: "Kitchen 1"
  command_topic: "K-LEDS/channel/0/set"
  payload_off: "OFF"
  brightness_command_topic: "K-LEDS/channel/0/set"
  brightness_state_topic: "K-LEDS/channel/0"
  on_command_type: "brightness"
  #availability_topic: "K-LEDS/status"
  #payload_available: "1"
  #payload_not_available: "0"

- platform: mqtt
  name: "Kitchen 2"
  command_topic: "K-LEDS/channel/1/set"
  payload_off: "OFF"
  brightness_command_topic: "K-LEDS/channel/1/set"
  brightness_state_topic: "K-LEDS/channel/1"
  on_command_type: "brightness"
  #availability_topic: "K-LEDS/status"
  #payload_available: "1"
  #payload_not_available: "0"

Let's see how it works

peekayv, 2018-11-16:
Any updates ?
Below is my configuration of seperate dimmer-channels on Quin, however switch is still common - turning one off / on, turns also the other one.
You could also use channel topic to turn it off, as you described, but then turning it on again would reset brightness level.
Did you succeed with DUMMY_RELAY_COUNT ?
```
  - platform: mqtt
    name: 'salon_tv'
    friendly_name: 'Swiatlo Salon TV'
    state_topic: 'ESP_SALON/relay/0'
    command_topic: 'ESP_SALON/relay/0/set'
    brightness: true
    brightness_scale: 255
    brightness_command_topic:  'ESP_SALON/channel/0/set'
    brightness_state_topic: 'ESP_SALON/channel/0'
    payload_on: 1
    payload_off: 0
  - platform: mqtt
    name: 'salon_dining'
    friendly_name: 'Swiatlo Salon Jadalnia'
    state_topic: 'ESP_SALON/relay/0'
    command_topic: 'ESP_SALON/relay/0/set'
    brightness: true
    brightness_scale: 255
    brightness_command_topic:  'ESP_SALON/channel/1/set'
    brightness_state_topic: 'ESP_SALON/channel/1'
    payload_on: 1
    payload_off: 0

```

zen, 2018-11-17:
I didn't try with this option to be honest, I'm just not using switch and emulation on/off with brightness

zen, 2018-11-25:
Yes, that's right. Alexa integration is also useless without it

closed 2021-10-03 by mcspr