xoseperez/espurna #1223: [Feature request] Selective sensor debugging/error output

vogelfreiheit, 2018-09-23:
I'm enabling support for different I2C devices and also use SSL for MQTT, so web support is disabled (I experienced resets and crashes to the point the module became unusable, this is with a POW R2).

Unfortunately, telnet terminals get hosed with output from the sensor debugging functions. It would be nice to:

- Either have a terminal command that logs/shows how many errors (and what the last error was) exist for a given sensor registered in the system.  (This is the best option and a small table in memory should hold very little memory hostage... three integers for the sensor ID, error code and another field at most, resolved during terminal access to the string counterparts as needed).
- Or have a flag we can enable to avoid outputting sensor messages to telnet, but send them everywhere else. Unfortunately the debugging API doesn't make this any easier.

My current workaround is an ifdef around the offending spots, but this is far from optimal.

xoseperez, 2018-09-23:
I understand what you mean. Sensor debug is probably the ugliest in ESPurna, that's why it has it's own build settings to enable/disable it.

Maybe a good approach would be to tag debug messages with the typical INFO, DEBUG, ERROR tags and to be able to set the debug level from the terminal...

vogelfreiheit, 2018-09-24:
Tagging sounds like a not too complicated way to go about it. We can just use a single toggle, maybe a bitmask/flag integer, so the increase in used storage is minimal. Then on runtime we can toggle sensor debugging on or off.

I have other questions that probably belong better over e-mail, to avoid cluttering the tracker. One of them is figuring out any compile time optimizations. I tested building with platform 150, 160, and 173, and 180, and I see significant differences in bin size, not for the better. It might be a good idea to run something on the codebase and detect code reuse. And then also figure out if we can compact things a little.

Is there a tool that can speak a standard protocol to espurna to essentially remove the need for the web ui entirely? I see the terminal implements quite a bit of the web available variables, but one major advantage of having a standalone remote API without the overhead of the web UI is that suddenly we can run TLS everywhere, as much of the memory is available and not hosed.

Also: any use cases for direct influxdb support? I haven't found what is used there.

ColinShorts, 2018-09-24:
If messages in the logs are tagged, a natural progression might be to allow remote syslog as an option too, although that might mean having to tweak quite a few bits of code. I'm not sure how many people would use it, but it might be quite useful for @vogelfreiheit  in this scenario?

xoseperez, 2018-09-25:
Remote syslog is already supported with the DEBUG_UDP_SUPPORT option when setting the DEBUG_UDP_PORT to 514.