xoseperez/espurna #1226: Question NODEMCU / TX RX

lblabr, 2018-09-23:
i'd like to use TX / RX - D9 / D10 as output, but it seems that there is some output on boot, sometimes the nodemcu does not boot with load on D9/D10

nay advice is welcome

thanks in advance

Lars

mcspr, 2018-09-25:
Output of debug messages? See DEBUG_PORT value. For D9/D10 it should be Serial1
https://github.com/xoseperez/espurna/blob/f01f2454cc51e6efa497bc4476660bb106d8fd32/code/espurna/config/general.h#L37-L39

lblabr, 2018-09-26:
i'd like to switch to GPIO Mode (Output) for D9/D10, as i read  Function_3, how do i do that ?

mcspr, 2018-09-27:
Have you tried using OUTPUT mode instead? `pinMode(D9, OUTPUT)` ([if you selected nodemcu as board](https://github.com/esp8266/Arduino/blob/master/variants/nodemcu/pins_arduino.h#L46))

DEBUG_PORT should be changed so it does not do Serial.begin(), which uses TX / RX (i'd suggest just replacing with Serial1 so tx can be read from the secondary). There's still chip boot message on original pin, which btw you can read on 74880 speed. That can't go away.