xoseperez/espurna #1227: Invalid characters in HASS configuration

robin13, 2018-09-24:
For the unit_of_measurement, the bare `%` is invalid:
```
sensor:
  - name: sonoff_th16_02_humidity
    platform: mqtt
    state_topic: sonoff-th16-02/humidity
    unit_of_measurement: %
```

Should be enclosed with quotes:

```
    unit_of_measurement: "%"
```

mcspr, 2018-09-25:
Regarding f01f245, there was previous issue with the same change https://github.com/xoseperez/espurna/commit/dfa732749111adf248450372a88f3c66b1681b6f at #850 

unit is wrongly quoted in discovery:
```
[010968] [MQTT] Sending homeassistant/sensor/espurnadev/config => {"name":"espurnadev_humidity","platform":"mqtt","state_topic":"test/espurnadev/humidity","unit_of_measurement":"\"%\""} (PID 12)
```

ha.config yaml:
```
sensor:
  - name: espurnadev_humidity
    platform: mqtt
    state_topic: test/espurnadev/humidity
    unit_of_measurement: "%"
```

I think quoting should happen at ha.config code instead.

*sidenote* There are sleep-inducing yaml spec rules of what can be without quotes and cannot (becoming yaml language directive instead of string):
http://yaml.org/spec/1.2/spec.html#id2772075
Easier read at camel docs, python yaml lib. ctrl+f `for a plain scalar to be a string, it must meet these restrictions`:
https://camel.readthedocs.io/en/latest/yamlref.html

xoseperez, 2018-09-25:
True. I changed that too quick and dirty.

closed 2018-10-07 by xoseperez