xoseperez/espurna #1228: Strange counting of Joules on POW2

JohnOH, 2018-09-24:
I am seeing an issue on the joules counter. My device has been ticking over for a few days now with the relay open. I have been monitoring the power consumption of what is presumable the processor in my Sonoff POW Rev 2. Below is a snip of the MQTT log the values are replicated on my ThingSpeak channel:
```
Sonoff1/host Sonoff1
Sonoff1/ip 192.168.1.45
Sonoff1/mac 84:F3:EB:B1:E4:70
Sonoff1/rssi -63
Sonoff1/uptime 156704
Sonoff1/datetime 2018-09-23 13:40:18
Sonoff1/freeheap 18008
Sonoff1/relay/0 0
Sonoff1/vcc 3264
Sonoff1/status 1
Sonoff1/loadavg 4
Sonoff1/current 0.000
Sonoff1/voltage 243
Sonoff1/power 0
Sonoff1/energy 35483         < steady rise to this point
Sonoff1/current 0.000
Sonoff1/voltage 244
Sonoff1/power 0
Sonoff1/energy 402754       < then aa irrational leap
Sonoff1/current 0.000
Sonoff1/voltage 244
Sonoff1/power 0
Sonoff1/energy 402770
Sonoff1/current 0.000
Sonoff1/voltage 243
Sonoff1/power 0
Sonoff1/energy 402786
Sonoff1/app ESPURNA
Sonoff1/version 1.13.2
Sonoff1/board ITEAD_SONOFF_POW_R2
Sonoff1/host Sonoff1
Sonoff1/ip 192.168.1.45
Sonoff1/mac 84:F3:EB:B1:E4:70
Sonoff1/rssi -64
Sonoff1/uptime 157004
Sonoff1/datetime 2018-09-23 13:45:18
Sonoff1/freeheap 21208
Sonoff1/relay/0 0
Sonoff1/vcc 3263
Sonoff1/status 1
Sonoff1/loadavg 1
Sonoff1/current 0.000
Sonoff1/voltage 243
Sonoff1/power 0
Sonoff1/energy 402802
Sonoff1/current 0.000
Sonoff1/voltage 244
Sonoff1/power 0
```
Might there be an issue withe the collection of a multi-byte number?

JohnOH, 2018-09-24:
```
            // Calculate energy
            unsigned int difference;
            static unsigned int cf_pulses_last = 0;
            unsigned int cf_pulses = _data[21] << 8 | _data[22];
            if (0 == cf_pulses_last) cf_pulses_last = cf_pulses;
            if (cf_pulses < cf_pulses_last) {
                difference = cf_pulses + (0xFFFF - cf_pulses_last) + 1;
            } else {
                difference = cf_pulses - cf_pulses_last;
            }
            _energy += difference * (float) _coefP / 1000000.0;
            cf_pulses_last = cf_pulses;
```
I am not much of an practitioner but the penultimate line has me wondering.

JohnOH, 2018-09-27:
I restarted my POW2 to check if the Energy reading went crazy at the same count. They didn't:
```
Sonoff1/datetime 2018-09-27 07:37:40
Sonoff1/freeheap 20704
Sonoff1/relay/0 0
Sonoff1/vcc 3264
Sonoff1/status 1
Sonoff1/loadavg 1
Sonoff1/current 0.000
Sonoff1/voltage 244
Sonoff1/power 0
Sonoff1/energy 39238
Sonoff1/current 0.000
Sonoff1/voltage 245
Sonoff1/power 0
Sonoff1/energy 39249
Sonoff1/current 0.000
Sonoff1/voltage 245
Sonoff1/power 0
Sonoff1/energy 39259
Sonoff1/current 0.000
Sonoff1/voltage 244
Sonoff1/power 0
Sonoff1/energy 39276
Sonoff1/current 0.000
Sonoff1/voltage 244
Sonoff1/power 0
Sonoff1/energy 39286
Sonoff1/app ESPURNA
Sonoff1/version 1.13.2
Sonoff1/board ITEAD_SONOFF_POW_R2
Sonoff1/host Sonoff1
Sonoff1/ip 192.168.1.45
Sonoff1/mac 84:F3:EB:B1:E4:70
Sonoff1/rssi -59
Sonoff1/uptime 173108
```

xoseperez, 2018-10-07:
I've been monitoring a couple of devices with a CSE7766 (a POW 2 and a S31) for the last 5 days and have not seen anything weird yet. I will keep this open in case you or me find something else that could explain the behaviour you have seen but I will remove it from this milestone.