xoseperez/espurna #1232: Configuration in #1231 results in + string added to the relay name for MQTT

vogelfreiheit, 2018-09-27:
Something is broken while building with platform 173 and 180, have not tested it with 160 but I will give it a go. It's very time consuming unfortunately.

This breaks Home Assistant, as + is a wildcard and won't allow me to add it to the configuration.  I am not quite sure what is going on, yet. + should actually match anything. But obviously from broker to device, if it needs a specific relay number, it will refuse to connect to the Espurna device, as HASS is complaining.

I haven't seen this behavior before, and the relay was correctly named as 0.

```
mqtt.reset
[861781] [MQTT] Disconnecting
[861788] [MQTT] TCP Disconnected
[861791] [MQTT] Disconnected!
[861792] +OK
[861796] [MQTT] Connecting to broker at BBB:VVV
[861798] [MQTT] Connecting as user esp-office
[861799] [MQTT] Using SSL
[861799] [MQTT] Client ID: ESPURNA-4AD772
[861799] [MQTT] QoS: 0
[861799] [MQTT] Retain flag: 1
[861800] [MQTT] Keepalive time: 300s
[861800] [MQTT] Will topic:XXX/status
[862081] [MQTT] Connected!
[862087] [MQTT] Unsubscribing to # (PID 1)
[862092] [MQTT] Subscribing to XXX/relay/+/set (PID 2)
[862097] [MQTT] Subscribing to XXX/pulse/+/set (PID 3)
[862103] [MQTT] Subscribing to XXX/led/+/set (PID 4)
[862108] [MQTT] Subscribing to XXX/action/set (PID 5)
[862120] [MQTT] Subscribe ACK for PID 2
[862124] [MQTT] Subscribe ACK for PID 3
[862126] [MQTT] Subscribe ACK for PID 4
[862146] [MQTT] Subscribe ACK for PID 5
```

vogelfreiheit, 2018-09-27:
Nevermind, after reading the code, it seems the relay code subscribes to the wildcard. Strangely, HASS is not able to detect the switch as active. I'll look further into it.

mcspr, 2018-11-01:
See https://www.eclipse.org/paho/files/mqttdoc/MQTTClient/html/wildcard.html
> For subscriptions, two wildcard characters are supported:
> - '#' character represents a complete sub-tree of the hierarchy and thus must be the last character in a subscription topic string, such as SENSOR/#. This will match any topic starting with SENSOR/, such as SENSOR/1/TEMP and SENSOR/2/HUMIDITY.
> - '+' character represents a single level of the hierarchy and is used between delimiters. For example, SENSOR/+/TEMP will match SENSOR/1/TEMP and SENSOR/2/TEMP.

Nonetheless, I see that you are using SSL, and that may have unexpected side-effects due to memory usage. HASS autodiscovery may not be sent out as a result. But MQTT would've still worked, did you actually tried debugging with something like mosquitto_pub / mosquitto_sub?

xoseperez, 2018-12-09:
Any update on this?

closed 2019-02-15 by stale[bot]