xoseperez/espurna #1233: Sonoff RF Bridge HASS config

vaidyasr, 2018-09-28:
The Sonff RF Bridge HASS configuration is not showing anything when I click 'Show' button. I can able to see on other esp8266 switches.
I made my own firmware with 16 switches enabled, compiled and uploaded. The RF switch configuration page for switch 10 to 16 contains 'Switch OFF' default value as 36EC01CC054610C10E.

xoseperez, 2018-10-05:
I see, the settings module is not handling properly the settings with indexes over 9. Will fix that. About the `Show` button, it could be a memory issue. Will check that too.

moddsson, 2018-10-18:
Is there technically anything that prevents one to define up to 24 indexes? - I configured my own with 24 switches and I find that the rfbridge resets completely and looses all settings.

moddsson, 2018-10-18:
Ok, I just saw that this is partially addressed in https://github.com/xoseperez/espurna/issues/1138

mcspr, 2018-12-20:
Looking at possible ways to "get" config, maybe it is worth to just drop it completely? Discovery is good enough with recent changes (like making discovered devices reconfigure or be deleted when removing discovery message)

Referencing #1300 again - I'd rather not send them separately (or workaround it using immediate pingback)
One improvement I can think of is showing available topics of the device i.e. render "mqttTopic" value and avoid looking at "mqtt.reset" output to figure out what device subscribes / sends

PR #1535 (2019-02-10): Print each HA config entry separately
Fixes #1233 
Does involve sort-of a hack by knowing that DEBUG_MSG will not print timestamp if the last symbol is not new-line/cr-return.
closed 2019-02-10 by xoseperez