xoseperez/espurna #1235: ITEAD_SONOFF_DUAL_R2

AlbertWeterings, 2018-09-28:
There is an issue with ITEAD_SONOFF_DUAL_R2 LED mode.

if LED mode is set to "Find me & WiFi" and relay 1 (Switch #0) is activated the LED turns off after a few seconds. It flashes on WiFi status. If relay 1 (Switch #0) is deactivated the LED turns on again. It looks to me the LED function is reversed based on position of relay 1. As there is a red/green LED to indicate the position of the relay's this is not needed.

I also found my ITEAD_SONOFF_DUAL_R2 not responding to anything after some hours not yet able to find if this is hardware related.

xoseperez, 2018-10-05:
The "Find me" mode means just that: turn LED on when the relay is OFF and vice-versa. The idea is to help you find a switch that controls a light in the dark. The "Find me & WiFi" mode will overlap the "WiFi" mode on top of it so it will briefly blink (on or off) every 5 seconds if connected to your home network.

The other issue is more worrisome. No button presses? Telnet?

AlbertWeterings, 2018-10-11:
The problem ITEAD_SONOFF_DUAL_R2 was hardware related I was having 2 out of 3 units with the same issue. It was caused by a electrolytic capacitor near the regulator not being soldered well. 

What I mentioned over the LED behavior is not something that needs to be fixed I noticed this reversing behavior of the LED as unnecessary as the doal color LED is showing relay state.

xoseperez, 2018-12-10:
Shall I close this, then?

xoseperez, 2018-12-10:
OK, I'm closing it.

closed 2018-12-10 by xoseperez