xoseperez/espurna #1239: "Clear counts" on rfm69 does not reset node count properly.

Trickx, 2018-10-01:
Issue: If the button "Clear counts" is pressed, the node count on the status page does not start to count anymore.

Reason: The reset code does not reset .count.
```
void _rfm69Clear() {
    for(unsigned int i=0; i<255; i++) {
        _rfm69_node_info[i].duplicates = 0;
        _rfm69_node_info[i].missing = 0;
    }
    _rfm69_node_count = 0;
    _rfm69_packet_count = 0;
}
```

Solution:

```
void _rfm69Clear() {
    //for(unsigned int i=0; i<255; i++) {
    for(unsigned int i=0; i<(sizeof(_rfm69_node_info)/sizeof(_node_t)); i++) {
        _rfm69_node_info[i].duplicates = 0;
        _rfm69_node_info[i].missing = 0;
        _rfm69_node_info [i].count = 0;
    }
    _rfm69_node_count = 0;
    _rfm69_packet_count = 0;
}
```

lobradov, 2018-10-01:
Good catch!
Care to raise a PR for a fix?

__
//_

Trickx, 2018-10-01:
I would, but my espurna-fork is currently being used for other features - not ready for PR.
That's why I posted the fixed code here.

lobradov, 2018-10-01:
You can always have a new branch :) 
Or even use online editor for small changes... 
__
//_

Trickx, 2018-10-01:
-> PR #1241

xoseperez, 2018-10-05:
Merged into `dev`. Thank you!

closed 2018-10-05 by xoseperez