xoseperez/espurna #1243: Wemos D1 mini & D3/GPIO0 as output

lblabr, 2018-10-04:
i defined GPIO 0 as output, it does not seem to work, does anybody has some hints ?

// 	Buttons
//    #define BUTTON1_PIN         0
//    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
// 	LEDs
//    #define LED1_PIN            2
//    #define LED1_PIN_INVERSE    1
//    #define BUTTON2_PIN         2
//    #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_LOW

	#define EVENTS_SUPPORT      1
	#define EVENTS_PIN          16
	#define EVENTS_TRIGGER      1
	#define EVENTS_INTERRUPT_MODE     RISING

    // Relays
    #define RELAY1_PIN          2
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL
    #define RELAY2_PIN          0
    #define RELAY2_TYPE         RELAY_TYPE_NORMAL
    #define RELAY3_PIN          4
    #define RELAY3_TYPE         RELAY_TYPE_NORMAL
    #define RELAY4_PIN          5
    #define RELAY4_TYPE         RELAY_TYPE_NORMAL

thanks in advance Lars

xoseperez, 2018-10-05:
There is no reason why this configuration shouldn't work. Are you sure the relay is on GPIO 0?

lblabr, 2018-10-05:
yes i´m sure. i also measured the voltage level on the pin it does not change from 3,3 V to 0 if i switch the relay from UI. 
Today i will test with another wemos, maybe there is something on the board

xoseperez, 2018-10-05:
GPIO0 usually has an external pull-up to enter normal mode on boot. You must pull it low to enter programming mode. But after that, you should be able to pull it low from software.

lblabr, 2018-10-05:
so far i understud, to define GPIO 0 as output should do that !?

xoseperez, 2018-10-05:
Yes. Other boards already use GPIO 0 for relay management, so there should be no problem with it...

lblabr, 2018-10-05:
i tried with second board, voltage level (D3) does not change with the above configuration

lblabr, 2018-10-05:
btw. its not orig. wemos

lblabr, 2018-10-05:
i did some measuremnt also an ESP itself the voltage does not change
do you have some hints how can i test whats wrong ?

lblabr, 2018-10-05:
https://phone-china.en.made-in-china.com/product/vqAnyKkWMdtO/China-Esp8266-Esp-12-Wemos-D1-Mini-WiFi-4m-Bytes-Development-Board-Module.html

lblabr, 2018-10-05:
adding:

pinMode(0,OUTPUT);
digitalWrite(0,LOW);

in Setting changed the voltage level to "0"

lblabr, 2018-10-05:
after adding:

pinMode(0,OUTPUT);

in setup procedure it works as expected... maybe the boundaries are somewhere wrong ... GPIO "**0**"

xoseperez, 2018-10-07:
Setting the pinMode of the relays is done in the _relayConfigure method:

https://github.com/xoseperez/espurna/blob/acf68ac5c8dbf09473b359d94792dda51201913b/code/espurna/relay.ino#L518-L531

lblabr, 2018-10-08:
i already have seen that, but it seem not to work... i will remove my additional pinMode in the end of the setup() and try again...

i use GPIO 0 and 2 as relay one of  relays is switching during boot precedure,, i think it shouldn´t

lblabr, 2018-10-09:
verified: 

GPIO 0 does not work as output without an additional 

pinMode(0,OUTPUT); 

at the end of the setup function, i did not tested somewhere in the middle

xoseperez, 2018-12-11:
It's got to be a hardware-related issue. I just tested with an (original) Wemos D1 mini and it works as expected without changes in the code.

lblabr, 2018-12-21:
how do i cover my device ? any suggestion ?

Misiu, 2019-01-13:
@lblabr I'm connecting 4 relays right now (using D0, D5, D6 and D7) pins because when I used D3 and D4 (GPIO0 and GPIO2) they turned on for a short time when boot.
Tomorrow I'll try uploading a custom build with a similar config You posted. I have original Wemos D1 and couple of cheap non-original.
@xoseperez did You have an issue with relays connected to D3, D4 or D8 (as I remember correct D3 and D4 must be high and D8 must be low to start)? Do You have an idea how this problem can be solved (if not by software then maybe hardware)?

closed 2019-03-22 by stale[bot]